Xilinx/Vitis-AI #434: vaitrace runmode: debug not working on ZCU 102 prebuild image

Fabioni, 2021-06-02:
Hi all,

I want to do fine-grained profiling of e.g. resnet50 on a ZCU 102 board with the prebuild image (xilinx-zcu102-dpu-v2020.2-v1.3.1.img.gz). I followed <https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Profiler>.

Following the instructions under 'Get Started with Vitis AI Profiler':  
Since I use the prebuild image, I do not have to do the `Preparing debug environment for vaitrace in MPSoC platform` and I do not have to do `Preparing debug environment for docker If you are using Vitis AI with docker` because I do not use it with docker when I use the ZCU102 right?

When I run the vaitrace command with runmode: debug, it won't run with that but in normal mode, as you can see the following picture (`INFO:root:VART will run xmodel in [NORMAL] mode`): 

[image]

the used `vaitrace_config.json` is:
```
{
  "options": {
      "runmode": "debug"
  }
}
```

Also in Vitis Analyzer I only see:
[image]

but what I wanted is:
[image]

Output of `lsb_release -a` is the following:
```
Distributor ID:	petalinux
Description:	PetaLinux 2020.2
Release:	2020.2
Codename:	zeus
```

-----

I think I folled all the instructions and have all the requirements so that it should work.  Maybe somebody can help me to find the problem or can confirm that this is a bug.

Greetings
Fabian

qianglin-xlnx, 2021-06-06:
Hi @Fabioni 
Please remove the following line (line121)  in /usr/bin/xlnx/vaitrace/vaitraceDefaults.py on the target and try it again. 

[image]

Fabioni, 2021-06-06:
> Hi @Fabioni
> Please remove the following line (line121) in /usr/bin/xlnx/vaitrace/vaitraceDefaults.py on the target and try it again.
> 
> [image]

Hi @qianglin-xlnx 

thanks a lot. That worked immediately. Kind of frustrating how easy it was after days of trying. Did I do or understand something wrong or is it just a bug or wrongly documented?

Greetings,
Fabian

Fabioni, 2021-06-07:
Okay,

I found the problem. In the file /usr/bin/xlnx/vaitrace/vaitraceDefaults.py in the dictionary traceCfgDefaule is written in my image

```python
traceCfgDefaule = {
    "runmodel": "normal",
    'collector': {},
    'trace': {

        "enable_trace_list": ["vitis-ai-library", "vart", "xnnpp_post_process", "custom"]
    },
    'control': {
        'cmd': None,
...
```

but in the image of a colleague is written the following. If I change m file to this, it works for me. (runmode inside options)

```python
traceCfgDefaule = {
    "options": {
        "runmodel": "normal"
    },
    'collector': {},
    'trace': {

        "enable_trace_list": ["vitis-ai-library", "vart", "xnnpp_post_process", "custom"]
    },
    'control': {
        'cmd': None,
...
```

AlaraWu, 2021-08-06:
> Okay,
> 
> I found something strange. In the file /usr/bin/xlnx/vaitrace/vaitraceDefaults.py in the dictionary traceCfgDefaule is written in my image the following. It's the same as in the picture @qianglin-xlnx sent.
> 
> ```python
> traceCfgDefaule = {
>     "runmode": "normal",
>     'collector': {},
>     'trace': {
> 
>         "enable_trace_list": ["vitis-ai-library", "vart", "xnnpp_post_process", "custom"]
>     },
>     'control': {
>         'cmd': None,
> ...
> ```
> 
> but in the image of a colleague, who acutally has the same image version is written the following.
> 
> ```python
> traceCfgDefaule = {
>     "options": {
>         "runmodel": "normal"
>     },
>     'collector': {},
>     'trace': {
> 
>         "enable_trace_list": ["vitis-ai-library", "vart", "xnnpp_post_process", "custom"]
>     },
>     'control': {
>         'cmd': None,
> ...
> ```
> 
> See that **runmodel** (with L) is inside options and not _runmode_ (without L) outside options.

Hi, my vaitraceDefaults.py file is the same as your colleague. I use the pre-built image v1.3.1 as well and now my profiling result turns out to be only two rows, the same as your previous problem. Do you know how to solve this? I tried to remove the `"runmodel": “normal”` but it obviously didn't work well.

li900309, 2022-03-10:
hi, the "runmode" in config file is discarded since vai1.4, now we have a simple way to control it , just add **--fine_grained** after vaitrace it will start fine-grained profiling. https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Profiler#command-line-usage

closed 2022-03-10 by qianglin-xlnx

gunertatar, 2022-11-27:
Hi;
I m using Kria KV260 vision board and i executed the tf_resne50 example from the Vitis AI VART. I d like to look at the performance of behavior of the board and used the vaitrace ./resnet50 resnet50.xmodel. After execution i got only " profile_summry.csv, summary.csv, vart_trace.csv, vitis_ai_profile.summary, xrt.run_summary" files. However, when i look around i think, xclbin.ex.run_summary file also should be in the executed file. By the way Vitis unifed software platform and vitis_analyzer version 2021.1.
Is there anyone could help me please? I really need to check all files belongs to execution such as HAL_API Calls, OpenCL data, Data Transfer, DPU, buffer exchange and so on.

Thanks