Xilinx/Vitis-AI #435: DPU-TRD custom petalinux platform with 3 DPU-Cores problems

JerrySciangula, 2021-06-04:
Hi everyone,
I'm a research fellow at Scuola Superiore Sant'Anna in Pisa, Italy.

I tried to create a custom hardware DPU v3.3 configuration starting from the DPU-TRD tcl scripts, since for my application I need 3 DPU cores, running three different models.

Here you have my configuration of the trd_prj.tcl file:
_dict set dict_prj dict_param  DPU_CLK_MHz            {100}
dict set dict_prj dict_param  DPU_NUM                {3}
dict set dict_prj dict_param  DPU_ARCH               {4096}
dict set dict_prj dict_param  DPU_RAM_USAGE          {high} 
dict set dict_prj dict_param  DPU_CHN_AUG_ENA        {1}
dict set dict_prj dict_param  DPU_DWCV_ENA           {0}
dict set dict_prj dict_param  DPU_ELEW_MULT_ENA      {0}
dict set dict_prj dict_param  DPU_AVG_POOL_ENA       {1}
dict set dict_prj dict_param  DPU_CONV_RELU_TYPE     {2}
dict set dict_prj dict_param  DPU_SFM_NUM            {0}
dict set dict_prj dict_param  DPU_DSP48_USAGE        {high}
dict set dict_prj dict_param  DPU_URAM_PER_DPU       {0}
dict set dict_prj dict_param  REG_CLK_MHz            {100}
dict set dict_prj dict_param  HP_CLK_MHz             {334}
dict set dict_prj dict_param  CN_STGY                {flat}
dict set dict_prj dict_param  DPU_IP_FOLDER          {dpu}
dict set dict_prj dict_param  DPU_SAXICLK_INDPD      {1}
dict set dict_prj dict_param  DPU_CLK_GATING_ENA     {1}
dict set dict_prj dict_param  DPU_DSP48_MAX_CASC_LEN {4}
dict set dict_prj dict_param  DPU_TIMESTAMP_ENA      {1}_

The project is correctly synthesized and implemented with Vivado 2020.2, generating successfully the bitstream and the exported hardware platform. At this point, I managed to create a petalinux image through Petalinux Tools 2020.2, as suggested by the DPU-TRD Vivado flow.

Here, something strange happens: After compiling custom DNN models using the arch.json file of the new generated platform, the software application is executed, (it successfully runs on the standard Vitis AI 1.3.2 petalinux image with 2 DPU cores), reporting that the models have different fingerprint from that of the DPU, as you can see from the following image: 

[image]

Therefore, I tried to recompile the models using that DPU fingerprint, but another error arised, reporting a failure of the DPU execution:

[image]

Does anyone have an idea of what is happening?
Can anyone help me to try to understand these issues?

Thanks in advance.

qianglin-xlnx, 2021-06-09:
Hi @JerrySciangula 
Could you update the VART to 1.3.2 and try it again? 
Find the package in https://www.xilinx.com/bin/public/openDownload?filename=vitis-ai-runtime-1.3.2.tar.gz

JerrySciangula, 2021-06-09:
Hi @qianglin-xlnx,
thank you for your reply. I tried with the Vitis AI Runtime 1.3.2 and it worked.

However, the three custom models does not run on the 3 DPU cores, since just 2 cores are used, as you can see from the picture below:

[image]

I could not trace the execution with vaitrace command, since the custom petalinux image does not support the XRT environment used by the command.

Is there any methods to map a model to a specific core?
Is it possible modify the VART code to execute the three different models on three different DPU cores?

qianglin-xlnx, 2021-06-09:
Hi @JerrySciangula 
You can create multi runners to use different DPU cores. One runner will bind one core,  multiple runners will use polling to allocate cores. 
For example  runner0 ->  core0  ,  runner1 -> core1 , runner2-> core2   ,  runner 3 -> core0 ,runner4 -> core1 , runner 5-> core2 , runner6 -> core0 …

JerrySciangula, 2021-06-09:
Hi @qianglin-xlnx, 
thank you for the advice, but I already have 3 different runners, initialized with three different model subgraphs, in my code, but still running on just 2 DPU cores and not on 3.

I can share here the main file:

[main.zip](https://github.com/Xilinx/Vitis-AI/files/6624823/main.zip)

I cannot figure out how to solve this issue.
Thank you for the patience.

qianglin-xlnx, 2021-06-09:
Could you run any of multi-thread samples in demo/Vitis-AI-Library to test if the third DPU core work? Such as the following sample running in 4 threads. After the program runs, check the status of DPU again.

root@xilinx-zcu102-2020_2:~/Vitis-AI/demo/Vitis-AI-Library/samples/classification# ./test_performance_classification resnet50 test_performance_classification.list -t 4

JerrySciangula, 2021-06-11:
Hi @qianglin-xlnx, 
I just tried the example you told me. It runs on all the 3 DPU-cores! Still, I can't figure out why my code with 3 threads and 3 different runners does not use all the cores.

lishixlnx, 2021-07-20:
The code attached is incomplete.
can you please do below test:
run any simple example in vitis-ai-library in loop mode, ensure it occupy one dpu.
then run your code again to see if it uses the dpu that not used before.

qianglin-xlnx, 2022-02-22:
HI @JerrySciangula 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-22 by qianglin-xlnx