Xilinx/Vitis-AI #438: pytorch框架下的TraDes网络量化没有问题，使用zcu104的dpu编译报错，使用VCK5000编译没有问题。这是什么原因

xiehou123456, 2021-06-09:
您好，我在编译自己网络的时候出现以下问题，请求大佬帮忙解决，使用zcu104的dpu编译报错，使用VCK5000编译没有问题。
[image]

qianglin-xlnx, 2021-06-09:
Hi @xiehou123456 
VCK5000和zcu104 对应的DPU IP 不同，不同的DPU IP其特性、限制等都不一样。因此编译器在编译模型时，会根据对应的DPU 进行分析、判别，如遇到DPU 所不支持的特性，或者参数不在支持范围内，则会出现编译失败的情况。
具体编译失败的原因，还需要您发模型给我们分析下。谢谢。

xiehou123456, 2021-06-10:
hi @qianglin-xlnx 
这是我们的模型，麻烦您帮忙看一下，谢谢
[DLASegv0_int.xmodel.zip](https://github.com/Xilinx/Vitis-AI/files/6627999/DLASegv0_int.xmodel.zip)

qianglin-xlnx, 2021-06-15:
Hi @xiehou123456 
Op_name 为DLASegv0__DLASegv0_DLAUp_dla_up__IDAUp_ida_0__Sequential_node_1__Conv2d_0__input_116的conv，
Input channel 为257，weights的channel却是512.  模型似乎不太对，能否把原始模型发下，以便我们进一步分析。谢谢。

xiehou123456, 2021-06-16:
hi @qianglin-xlnx 
谢谢你帮忙分析，这些是我生成的量化文件和该项目的模型压缩包
[quantize_result.zip](https://github.com/Xilinx/Vitis-AI/files/6659745/quantize_result.zip)

[model.zip](https://github.com/Xilinx/Vitis-AI/files/6659764/model.zip)

xiehou123456, 2021-06-16:
hi @qianglin-xlnx 
这个是我原始模型，感谢您查看
[Uploading model_last_docker_without_detach.pth…]()

qianglin-xlnx, 2021-06-16:
@xiehou123456 
能否发一下你们量化用的脚本？多谢。

xiehou123456, 2021-06-16:
@qianglin-xlnx 
您好，我们是直接在运行的trades源码中的detector脚本中507～547行添加量化代码的（其中527和529页分别注释跑）
[detector.py.zip](https://github.com/Xilinx/Vitis-AI/files/6660598/detector.py.zip)

niuxjxlnx, 2022-03-14:
Hi,

There is a bug for depthwise transpose conv2d handling in pytorch quantizer in vitis-ai 1.4, which has been fixed in Vitis-AI 2.0.

closed 2022-03-14 by niuxjxlnx