Xilinx/Vitis-AI #439: xdputil shows -  No module named 'tools_extra_ops'

Maharshi-Aupera, 2021-06-09:
Hi Xilinx Community,

I have been trying to debug the xmodel and use tools provided by Xilinx here: https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Library/usefultools

However, when I try to use `xdputil` is first shows the numpy package missing related error then it shows this `No module named tools_extra_ops`. 

Is a script with this name `tools_extra_ops.py` missing from the repository or I am running this on a wrong device and with a wrong conda environment? Please suggest how to solve this.

Thanks.

Maharshi-Aupera, 2021-06-09:
I think I found them it is in the `src/` directory, do I need to build it?

qianglin-xlnx, 2021-06-09:
Hi @Maharshi-Aupera 
An easy way to use xdpuil is to install the VART1.3.2. After the installation, you can directly run xdputil. 
Note that, xputil is not supported in Vivado flow. It will be fixed in VAI1.4.

Maharshi-Aupera, 2021-06-09:
Thanks @qianglin-xlnx, I will give it a try and close the issue asap.

closed 2021-06-10 by Maharshi-Aupera