Xilinx/Vitis-AI #441: Understanging dpu_task APIs for input data tensor

Maharshi-Aupera, 2021-06-10:
Hi Xilinx Community,

I read through the APIs from `dpu_task.hpp` file and also tried understanding `dpu_task_imp.cpp` file. I was just wondering about the difference between `setInputDataArray()` and `getInputTensor()`?

My understanding is that the `setInputDataArray()` can be used as a replacement of the `setMeanScaleBGR()`  and `setImageBGR()` but then the pre-processing needs to be done on the user end, if needed.

For the `getInputTensor()`, I think this is useful when we have 2 kernels (DPU sub-graphs) in one xmodel. We read this tensor from the first kernel and then set some required values to this vector that we got from `getInputTensor()` API. Lastly, we just calll `task->run(1u)` for the second kernel and we should get the results. Here, the `task` is a pointer to the `DpuTask`.

Is my understanding correct on the two aforementioned API usages? If not, please correct me. 

Thanks.

lishixlnx, 2021-07-20:
Your understanding is correct.

Maharshi-Aupera, 2021-07-20:
Thanks @lishixlnx.

closed 2021-07-20 by Maharshi-Aupera