Xilinx/Vitis-AI #444: arch.json for ultra96 v2 embedded FPGA board

JianmingTONG, 2021-06-12:
Hi 

I'm running the model compilation using Vitis AI for a resnet18 finetuned model. However, the arch.json of the ULTRA96 v2 embedded board under the path ```/opt/vitis_ai/compiler/arch``` in the tutorial is not given. Might I ask for the help to get the arch.json for ultra96 v2?

Best
JT

JianmingTONG, 2021-06-14:
I found a list of available values for DPU on the website (https://forums.xilinx.com/t5/AI-and-Vitis-AI/vitis-ai-1-3-with-ultra96/td-p/1189251) as follows:
```
DPUCAHX8H_ISA2=>0x20200000000002a,
DPUCAHX8H_ISA2_ELP2=>0x20200000000002e,
DPUCAHX8L_ISA0=>0x30000000000001d,
DPUCVDX8G_ISA0_B16384C64B1=>0x600000076080812,
DPUCVDX8G_ISA0_B8192C32B1=>0x600000076080811,
DPUCVDX8G_ISA0_B8192C32B1_ELP4=>0x600000076040411,
DPUCVDX8G_ISA0_B8192C32B3=>0x600000076080831,
DPUCVDX8G_ISA0_B8192C32B3_DW=>0x6000000f6088831,
DPUCVDX8G_ISA0_B8192C32B3_I4W8B2=>0x600000276080831,
DPUCVDX8G_ISA0_B8192C32B3_I8W4B2=>0x600000376080831,
DPUCVDX8G_ISA0_B8192C32B3_I8W8B2=>0x600000176080831,
DPUCVDX8H_ISA0=>0x5000000000007ee,
DPUCZDI4G_ISA0_B4096_DEMO_SSD=>0x400002003220206,
DPUCZDI4G_ISA0_B8192D8_DEMO_SSD=>0x400002003220207,
DPUCZDX8G_ISA0_B1024_MAX=>0x1000020f7014402,
DPUCZDX8G_ISA0_B1024_MIN=>0x100002022010102,
DPUCZDX8G_ISA0_B1152_MAX=>0x1000020f7012203,
DPUCZDX8G_ISA0_B1152_MIN=>0x100002022010103,
DPUCZDX8G_ISA0_B1600_MAX=>0x1000020f7014404,
DPUCZDX8G_ISA0_B1600_MIN=>0x100002022010104,
DPUCZDX8G_ISA0_B2304_MAX=>0x1000020f7014405,
DPUCZDX8G_ISA0_B2304_MAX_BG2=>0x1000020f6014405,
DPUCZDX8G_ISA0_B2304_MIN=>0x100002022010105,
DPUCZDX8G_ISA0_B3136_MAX=>0x1000020f7014406,
DPUCZDX8G_ISA0_B3136_MAX_BG2=>0x1000020f6014406,
DPUCZDX8G_ISA0_B3136_MIN=>0x100002022010106,
DPUCZDX8G_ISA0_B4096_MAX=>0x1000020f7014407,
DPUCZDX8G_ISA0_B4096_MAX_BG2=>0x1000020f6014407,
DPUCZDX8G_ISA0_B4096_MAX_EM=>0x1000030f7014407,
DPUCZDX8G_ISA0_B4096_MIN=>0x100002022010107,
DPUCZDX8G_ISA0_B512_MAX=>0x1000020f7012200,
DPUCZDX8G_ISA0_B512_MIN=>0x100002022010100,
DPUCZDX8G_ISA0_B800_MAX=>0x1000020f7012201,
DPUCZDX8G_ISA0_B800_MIN=>0x100002022010101}
``

And based on this file located at ```./DPU_PYNQ/boards/Ultra96/dpu_conf.vh```, I think for ultra96 v2 it might need to choose 
```DPUCZDX8G_ISA0_B1600_MAX``` as the arch.json, so I specify the arch.json of ultra96 v2 as follows

```
{
    "target": "DPUCZDX8G_ISA0_B1600_MAX"
}
```

hoadv-qh, 2021-07-30:
Hi @JianmingTONG 

>{
>"target": "DPUCZDX8G_ISA0_B1600_MAX"
>}

Does it work?
By the way, what kind of Linux image you use on Ultra96v2 board?

qhy991, 2022-02-09:
@hoadv-qh After my practice for mores than 5 hours, the fingerprint of Ultra96v1 is DPUCZDX8G_ISA0_B2304_MAX_BG2=>0x1000020f6014405
Moving to tears for myself

hoadv-qh, 2022-02-18:
@qhy991 I haven't worked with Vitis AI & Ultra96V2 for several months.
I don't remember much about how to.

Could you describe your way to get fingerprint?
I think it will help other members whey they come to this topic.

Currently, I think command below can provide helpful info also.
<pre>
dexporer --whoami
</pre>

The output is something like below.
<pre>
DPU Core Configuration List]
DPU Core                : #0
DPU Enabled             : Yes
DPU Arch                : B4096
DPU Target Version      : v1.4.1
DPU Freqency            : 300 MHz
Ram Usage               : High
DepthwiseConv           : Enabled
DepthwiseConv+Relu6     : Enabled
Conv+Leakyrelu          : Enabled
Conv+Relu6              : Enabled
Channel Augmentation    : Enabled
Average Pool            : Enabled
</pre>

qianglin-xlnx, 2022-03-28:
Hi @JianmingTONG 
Is this issue solved?

qianglin-xlnx, 2022-04-02:
Hi @JianmingTONG 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-04-02 by qianglin-xlnx