Xilinx/Vitis-AI #446: file does not exist /home/root/vitis-ai_v1.2_dnndk_sample/resnet50 ./resnet50

rachelglenn, 2021-06-14:
The file resnet50 does not exit in the downloaded tar file, mentioned in the readme.md 

https://github.com/Xilinx/Vitis-AI/blob/v1.2/mpsoc/README.md

wanghy-xlnx, 2021-06-16:
Hi @rachelglenn ,
The resnet50 file is an executable binary file obtained after compiling the sample. You can prepare the cross-compilation environment according to [README.md](https://github.com/Xilinx/Vitis-AI/blob/v1.2/mpsoc/README.md) and execute `./build.sh zcu102` to get the executable file. Or install vitis-ai_v1.2_dnndk.tar.gz on the target board, and then use the `./build.sh zcu102` command to get the executable file.

qianglin-xlnx, 2022-02-16:
Hi @rachelglenn 
I wonder if you have solved this issue. If yes, could you please close this issue. If not, please also let me know. 

If we do not receive any reply from you within 1 weeks, we will assume that this issue has been resolved and will close this issue. Thank you.

closed 2022-02-25 by qianglin-xlnx