Xilinx/Vitis-AI #447: Convert the Alveo darknet yolo example to run on the ZCU102/ZCU104

rachelglenn, 2021-06-14:
I am trying to convert the darknet yolo example to run on a mpsoc. The example is located here.
https://github.com/Xilinx/Vitis-AI/tree/v1.2/alveo/apps/yolo/darknet_to_caffe

I have been looking at the code trying to understand how to run this example on a mpsoc(ZCU102 or ZCU104) and have not had luck figuring it out. Do you have any insight to help me out?

Thank you,

-Rachel

rachelglenn, 2021-06-16:
I am trying to get the carknet_to_cafffe example to work on an mpsoc board. How do I install the vai python libraries to run the script test_detect.sh? It imports vai. 
        from vai.dpuv1.rt import xdnn, xdnn_io
        from vai.dpuv1.rt.vitis.python.dpu.runner import Runner
https://github.com/Xilinx/Vitis-AI/blob/v1.2/alveo/apps/yolo/test_detect_vitis.py

Thank you!

jimheaton, 2021-06-16:
For MPSoC, you can use the yolov4 tutorial which covers how to do Darknet to Caffe conversion.
Yolov3 will convert directly, but Yolov4  will requires model modifications which is described in the tutorial:
https://github.com/Xilinx/Vitis-Tutorials/tree/master/Machine_Learning/Design_Tutorials/07-yolov4-tutorial

qianglin-xlnx, 2022-03-31:
Hi @rachelglenn 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-03-31 by qianglin-xlnx