Xilinx/Vitis-AI #448: Issue with Vitis AI Profiler (No such file or directory: '/sys/kernel/debug/tracing/trace_clock')

Luc-Meunier, 2021-06-15:
Hi,

I am trying to use the Vitis AI Profiler following the tutorial https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Profiler.
I am using an Alveo U280, and Vitis AI Docker container for CPU.

The following command causes the issue "**FileNotFoundError: [Errno 2] No such file or directory: '/sys/kernel/debug/tracing/trace_clock'**".
`sudo vaitrace ./resnet50 /usr/share/vitis_ai_library/models/resnet50/resnet50.xmodel
`

[image]

I am new to Linux and Vitis and I don't know how to solve this issue...

Thank you for your help.
Best regards,

Luc

qianglin-xlnx, 2021-06-15:
Hi @Luc-Meunier 
Could you please check the following?
[image]

Luc-Meunier, 2021-06-15:
Hello,
Thanks for helping me.

I added the line `-v /sys/kernel/debug:/sys/kernel/debug --privileged=true \` in the _docker_run.sh_ file.

[image]

However, I still get the error message **ERROR:root:This tool need run as 'root'**.
And I get an **std::runtime error** :

[image]

karan18205, 2021-06-17:
Facing same issue. It says Xilinx_xrt not set. Anyone any leads?

wilderfield, 2021-06-17:
source /opt/xilinx/xrt/setup.sh

^ this script is responsible for setting that environment variable. Ideally it would have been set for you.

karan18205, 2021-06-18:
Yes, solved it thanks.

Luc-Meunier, 2021-06-18:
Thank you, but unfortunately I still can't run _vaitrace_ (still XILINX_XRT not set despite sourcing the script).
Could you tell me what am I missing ?

[image]

SuZhan0322, 2022-09-01:
The latest VAI release has fixed this issue, please reopen this issue if there any other problems. thanks！

closed 2022-09-01 by SuZhan0322

hocken-li, 2023-07-28:
> The latest VAI release has fixed this issue, please reopen this issue if there any other problems. thanks！

My VAI is 3.0. Aslo source /opt/xilinx/xrt/setup.sh
But still have the problem, how to solve it?

SuZhan0322, 2023-07-30:
Hi @hocken-li , Can you share your steps and error message. Thanks!