Xilinx/Vitis-AI #450: dpu_yolov4 kernal crashing

daivikbhatia, 2021-06-17:
Hi,
I'm converting a yolo-v4 darknet model into tensorflow. The conversion, quantization and compilation was sucessful. I got the elf file and the kernal info. The kernal name in kernal info is "dpu_yolov4" but as soon as I put this kernal name in n2cube.dpuLoadKernal("dpu_yolov4"), the jupyter kernal of pynq 2.5 image crashes. Can you help me resolve it?
Also I tried same darknet model for caffe with kernal name "yolov4_voc" and everything ran smoothly.
I don't think the issue is related to elf file. I think it's n2cube library issue as loading kernal name is 1st step that does not require any model file.
the deployment reference I took was from here
https://github.com/Xilinx/Vitis-AI/blob/1.1/demo/DNNDK/tf_yolov3_voc_py/tf_yolov3_voc.py

zackdilan, 2021-06-18:
Can you show the Jupyter kernel log ??

daivikbhatia, 2021-06-21:
Sure here is a screenshot of logs.

[image]

daivikbhatia, 2021-06-21:
Here is a screenshot of kernal log

[image]

zackdilan, 2021-06-21:
Hi,

Thanks, just wanted to know! Have you already troubleshot the error showing in the log?

daivikbhatia, 2021-06-21:
Hi, 
No I'm still stuck here. Do you have any solution?

zackdilan, 2021-06-21:
I m sorry I cannot help you for the time being(I should find some time to set up DPU-PYNQ). But just make sure you can run the inference on the DPU image(not pynq) using regular dnndk runtime from Xilinx(To sanity-check the dpu kernel you got). There are demos on using dnndk APIs to create the inference application.