Xilinx/Vitis-AI #457: vai_c_xir [UNILOG][FATAL][XCOM_PM_FAIL][The compiler occurs an error when generating instructions, please contact us.]

Fabioni, 2021-06-26:
Hello,

when I use vai_c_xir to compile a xmodel of YoloV4-CSP I got from vai_q_pytorch, I get the following error. The folder where the log file should be `The compiler log will be dumped at "/tmp/vitis-ai-user/log/xcompiler-20210626-065906-2010"` is empty but exists.

[image]

I get three files but do not understand what they show:
`debug_bankstatus.txt`, `debug_depens.txt`, `debug_swimpool.txt`

The strange thing is, that when I leave the MaxPool layer with the unsupported size 9 and 13, the compiling works with just a warning:
[image]

Fabioni, 2021-06-26:
Now it worked

[image]

I followed https://github.com/Xilinx/Vitis-AI/issues/346#issuecomment-808292176

```
6.1 If not done so already, pull version 1.3.411 of the docker container with the following command:

docker pull xilinx/vitis-ai:1.3.411
6.2 Launch version 1.3.411 of the Vitis-AI docker from the Vitis-AI directory:

$ cd $VITIS_AI_HOME
$ sh -x docker_run.sh xilinx/vitis-ai:1.3.411
```

Before that i used `./docker_run.sh xilinx/vitis-ai-cpu:latest`
I quantized with lates and now compiled with `xilinx/vitis-ai:1.3.411` and it worked.

closed 2022-02-25 by qianglin-xlnx

sudharsan2000, 2022-03-23:
@Fabioni do you get any error regarding unregistered operators? Specifically for the `aten::silu` (which I guess is from [this](https://pytorch.org/docs/stable/generated/torch.nn.SiLU.html#torch.nn.SiLU) function)

Here is the complete stack trace:

```
[UNILOG][INFO] The compiler log will be dumped at "/tmp/vitis-ai-user/log/xcompiler-20220323-044739-141"
[0;31m[UNILOG][FATAL][XIR_UNREGISTERED_OP][Unregistered operator!] aten::silu_
[m*** Check failure stack trace: ***
This program has crashed!
Aborted (core dumped)
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
```