Xilinx/Vitis-AI #459: Unsupported op: type: expanddims

davidirobinson, 2021-07-01:
Currently is seems as though the `tf.expand_dims` is unavailable when running `vai_c_tensorflow`.

It's not listed as as "Operators Supported by TensorFlow" in https://www.xilinx.com/html_docs/vitis_ai/1_3/compiling_model.html#gyh1606713524607

Note that `tf.squeeze` appears to available, so it's a bit puzzling as to why the complementary expand function is not. 

```
Debug Log (using `vitis-ai-tensorflow` conda env):
[INFO] parse raw model     : 28%|██▊       | 223/805 [00:00<00:00, 25590.42it/s]             
[INFO] Namespace(inputs_shape=None, layout='NHWC', model_files=['mask_rcnn_R_50_FPN_1x.pb'], model_type='tensorflow', out_filename='./build/compile_zcu104/mask_rcnn_R_50_FPN_1x_org.xmodel', proto=None)
[INFO] tensorflow model: mask_rcnn_R_50_FPN_1x.pb
Traceback (most recent call last):
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/bin/xnnc-run", line 33, in <module>
    sys.exit(load_entry_point('xnnc==1.3.0', 'console_scripts', 'xnnc-run')())
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/__main__.py", line 194, in main
    normal_run(args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/__main__.py", line 178, in normal_run
    in_shapes=in_shapes if len(in_shapes) > 0 else None,
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/xconverter.py", line 131, in run
    xmodel = CORE.make_xmodel(model_files, model_type, _layout, in_shapes)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/core.py", line 104, in make_xmodel
    model_files, layout, in_shapes=in_shapes, model_type=model_t
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 97, in to_xmodel
    model_name, raw_nodes, layout, in_shapes, model_fmt, model_type
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 161, in create_xmodel
    xmodel = cls.__create_xmodel_from_tf1(name, layers, layout, in_shapes)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 267, in __create_xmodel_from_tf1
    xmodel_name, layout, layers, const_layer_dict, super_const_dict, in_shapes
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 3141, in __generate_xmodel
    f"*** Unsupported op: type: {op_type}, name: {op_name}"
ValueError: *** Unsupported op: type: expanddims, name: fpn/upsample_lat5/ExpandDims
```

qianglin-xlnx, 2021-07-22:
HI @davidirobinson 
Since this op is not supported by xcompiler, could you replace this op with another op, such as reshape? Thank you.

zhenzhen-Xilinx, 2022-03-07:
This problem can be solved, please use the latest version.

zhenzhen-Xilinx, 2022-03-15:
Hi @davidirobinson ,
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-03-15 by qianglin-xlnx