Xilinx/Vitis-AI #461: Missing .prototxt file after compilation and my md5sum.txt file doesn‘t have information of  .prototxt

littlemww, 2021-07-04:
I'm trying to run test_accuracy_yolov3_voc on the ZCU104 but it aborts because of a missing .prototxt file.

./test_accuracy_yolov3_voc 2007_test_gt.txt yolov4_person_pred.txt

[libprotobuf ERROR google/protobuf/text_format.cc:324] Error parsing text-format vitis.ai.proto.DpuModelParamList: 1:5: Message type "vitis.ai.proto.DpuModelParamList" has no field named "name".

WARNING: Logging before InitGoogleLogging() is written to STDERR
F0610 13:48:07.609874 1187 configurable_dpu_task_imp.cpp:140] Check failed: ok cannot parse config file. config_file=/usr/share/vitis_ai_library/models/yolov4_person/yolov4_person.prototxt
*** Check failure stack trace: ***

But I have copy the original yolov4_person.prototxt (Generated when I run the script ：bash scripts/darknet_convert.sh )to the board.
[image]

when I open the md5sum.txt file generated at compile time(I compiled according to the yolov4 tutorial, and there were no errors when compiling) I found that there was only one line inside：
[image]

but I see other md5sum.txt in the model are as follows:
[image]

So Is it because of the lack of information of prototxt in this file？If so, how to generate the correct md5sum.txt file at compile time？

Thanks in advance

jimheaton, 2021-07-07:
Are you pointing to the correct model in the test_accuracy_yolov3_voc.cpp file?
You will have to modify the following line to use your model: auto yolo = vitis::ai::YOLOv3::create("yolov3_voc_acc", true);
It sound like maybe you did this, but something is not right?

Here's what you should do:
1. Create a directory where test_accuracy_yolov3_voc is called yolov4_person, and copy the .prototxt and .xmodel files there. 
    Alternately, you can copy the dir to/usr/share/vitis_ai_library/models/yolov4_person/
2. Change the line in test_accuracy_yolov3_voc.cpp to: auto yolo = vitis::ai::YOLOv3::create("yolov4_person", true);

littlemww, 2021-07-08:
> Are you pointing to the correct model in the test_accuracy_yolov3_voc.cpp file?
> You will have to modify the following line to use your model: auto yolo = vitis::ai::YOLOv3::create("yolov3_voc_acc", [...]

Thank you for your reply
 I did these steps correctly,then I ran the command:bash -x build.sh,this error still occurs when I run the test program.
I saw the meta.json file in yolov4_person_compiled folder，Its contents are as follows：
[image]
but when I see the  meta.json file in dpu_yolov4_compiled folder(Already exists in the downloaded folder 07-yolov4-tutorial)，Its contents are as follows：
[image]
It contains information about config_file,but I don't have it in the meta.json file I generated.
So Is it because of the lack of information of config_file in this file？If so, how to generate the correct meta.json file at compile time？

Thanks

jimheaton, 2021-07-08:
The meta.json file in the Tutorial is based off an older version of Vitis-AI and needs to be updated.  Yours looks correct. T

The prototxt file info will not be listed in the meta.json file. This is something you have to create, and needs to be in the same directory and have the same name as  the xmodel.

Do you have a  yolov4_person.prototxt file in the yolov4_person dir? 

You can probably reuse the provided .prototxt file from the tutorial if the number of classes and and anchor points are the same. You will need to change the model and kernel name to yolov4_person.

Darshcg, 2021-07-16:
Dear @jimheaton,

I am having a doubt on.prototxt 
Why do we need .prototxt file for every model for inference? What is its a necessity and how to create that file?
Can you please help me to understand the above? Would be a great help.

Thanks and Regards,
Darshan

zz002, 2022-02-21:
@littlemww 
Has the problem been resolved? Can you show me your yolov4_person.prototxt file?
md5sum.txt is the file generated when we batch build the model, it doesn't work when running the model

zz002, 2022-02-21:
@Darshcg 
Some additional parameter information is stored in the .prototxt file,
When using the existing model library, you can make a copy of the .prototxt of other similar models and modify it
For example, the threshold of nms is used in post-processing [](https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Library/xnnpp/src/yolov3.cpp#L165)
We configure 
`grep nms_threshold /usr/share/vitis_ai_library/models/yolov3_bdd/yolov3_bdd.prototxt in .prototxt
     nms_threshold: 0.45
`

qianglin-xlnx, 2022-03-01:
Hi @Darshcg 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-03-01 by qianglin-xlnx