Xilinx/Vitis-AI #462: pytorch nndct pytorch version

reddittoob, 2021-07-06:
Hi, I'm trying quantization my network model. 

However, I'm getting error with pytorch version.

In terminal, pytorch_nndct can run only pytorch v1.4, however, I have to pytorch 1.8 over because of my graphic card(RTX 3080Ti). 

Anyone else know about this problem?

ngoclinhle, 2021-07-23:
You can use pytorch 1.8 to train and save model with torch.save(..., _use_new_zipfile_serialization=False)
Then the checkpoint can be opened and quantized in pytorch1.4 with pytorch_nndct

niuxjxlnx, 2022-02-21:
Pytorch 1.7 version has been supported in Vitis-AI 1.4, and pytorch 1.8, 1.9 version have been supported in Vitis-AI 2.0.
Further more, solution to change pytorch versions among the supported version range is released, please refer to the related part about script replace_pytorch.sh in user guide.

hanxue, 2022-03-17:
Hi @reddittoob ,

From https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Quantizer/vai_q_pytorch#quick-start-in-docker-environment , have you tried using `/opt/vitis_ai/scripts/replace_pytorch.sh` to use PyTorch 1.8?

closed 2022-03-17 by niuxjxlnx

cpracticer, 2022-11-03:
@hanxue  hello,dear hanxue,I followed the steps in replace_pytooch.sh.
1.I launched the VITIS AI 1.4.
2.conda create -n $1 --clone vitis-ai-pytorch
3.pip uninstall -y torch torchvision
4.pip uninstall  -y pytorch_nndct
5.pip install -r requirements.txt
6.pip install -r requirements.txt
7.cd pytorch_binding 
python setup.py bdist_wheel -d ./
pip install ./pytorch_nndct-*.whl
I got a pytorch version I wanted .It also works fine.
But after I close it.I found it's missing.
So have I miss some steps?
Does
   """  mamba install -y python=3.6 --force-reinstall
        sudo rm -rf /scratch/*     """"
mean save the pytorch environement I create?
Or how can I get a stable vai_q_pytorch  with pytorch version I wanted