Xilinx/Vitis-AI #463: Check failed: subgraph_.size() == 1u (3 vs. 1) model should have one and only one dpu subgraph.

abdulazizm, 2021-07-06:
I am using petalinux os and trying to run tvm compiled model (deeplabv3 resnet 101) which has 3 subgraphs it seems. Should I need to change anything in my build procedure to prevent having 3 subgraphs? How to get rid of this? Any examples? 

```
root@xilinx-zcu104-2020_2:~/workspace/scripts/deeplabv3_resnet101# python3 run_pytorch_deeplabv3_resnet101_zynq_fps.py -f "" -t 3 --nb_tvm_threads 1
File /home/root/.tvm_test_data/data/cat.png exists, skip.
File /home/root/.tvm_test_data/data/imagenet1000_clsid_to_human.txt exists, skip.
transform_image_torchvision_NCHW torch.Size([1, 3, 224, 224])
WARNING: Logging before InitGoogleLogging() is written to STDERR
F0614 08:36:17.361477  2907 dpu_func.cpp:57] Check failed: subgraph_.size() == 1u (3 vs. 1) model should have one and only one dpu subgraph.
*** Check failure stack trace: ***
Aborted
root@xilinx-zcu104-2020_2:~/workspace/scripts/deeplabv3_resnet101# 
```

Noticed some workaround here for CPP with VART (https://github.com/Xilinx/Vitis-AI/issues/153). Not sure how to get it down with Python - TVM

_Originally posted by @abdulazizm in https://github.com/Xilinx/pyxir/issues/40#issuecomment-860513869_

lishixlnx, 2021-07-19:
do you follow the steps in https://tvm.apache.org/docs/deploy/vitis_ai.html?
is your model for cloud or for edge?
is it possible to attach your code here?

hoadv-qh, 2021-07-30:
Hi @abdulazizm,

You can generate svg/png figure to see how many subgraphs in your model, and which are assigned to DPU/CPU.
In case of your model/building causing multiple graphs, you needs to handle multiple graph.

In example of Xilinx, model is simple so mostly there are only 1 subgraph assigned to DPU, others are Input/Output which assigned to CPU. That's why they implemented example with guard of 1 subgraph for DPU.

To generate picture for your graph, see UG1414, chapter *DPU Debug with VART*.
<pre>
Use xir tool to generate a picture for displaying the network's structure.
    Usage: xir svg <xmodel> <svg>

Note: In Vitis AI docker environment, execute the following command to install the required library.
    sudo apt-get install graphviz
</pre>

Hi @lishixlnx,

>is your model for cloud or for edge?

I think @abdulazizm run on edge device, since the log shows zcu104.

AlaraWu, 2021-07-30:
Hi @hoadv-qh,
I also came across the problem. Most of examples offered by Xilinx are 1-subgraph case so I don't know what I can refer to... Do you know how to handle multiple subgraphs?

hoadv-qh, 2021-07-30:
Hi @AlaraWu,

>Do you know how to handle multiple subgraphs?

I don't know how much complicated of your graph/model.
But I think example below is good hint.

https://github.com/Xilinx/Vitis-Tutorials/tree/master/Machine_Learning/Feature_Tutorials/pytorch-subgraphs

lishixlnx, 2022-02-16:
hi @abdulazizm , is this issue solved now? can this thread be closed?

qianglin-xlnx, 2022-02-25:
Hi @abdulazizm 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-25 by qianglin-xlnx