Xilinx/Vitis-AI #465: ERROR: Unsupported op: type: scale, name: cnn/block1/bn0/FusedBatchNormV3/add

FarhanAhmad4473, 2021-07-07:
I was able to create a graph using tensorflow and keras and quantize it as suggested by the Vitis AI documentation. During quantization my batch normalization layers are converted to scale operations. These operations fail during xmodel compilation. Since scale is probably just a multiply operation I am not entirely sure why the toolchain fails here or how I can debug this issue.

```
vai_c_tensorflow --frozen_pb="./quantize/deploy_model.pb" --arch /opt/vitis_ai/compiler/arch/DPUCAHX8H/U280/arch.json --output_dir="compile" --options="{'mode':'normal'}" --net_name "point_pillars"
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(inputs_shape=None, layout='NHWC', model_files=['./quantize/deploy_model.pb'], model_type='tensorflow', out_filename='compile/point_pillars_org.xmodel', proto=None)
[INFO] tensorflow model: quantize/deploy_model.pb
[INFO] parse raw model     :  3%|██▍                                                                        | 2/62 [00:00<00:00, 12427.57it/s]
Traceback (most recent call last):
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/bin/xnnc-run", line 33, in <module>
    sys.exit(load_entry_point('xnnc==1.3.0', 'console_scripts', 'xnnc-run')())
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/__main__.py", line 194, in main
    normal_run(args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/__main__.py", line 178, in normal_run
    in_shapes=in_shapes if len(in_shapes) > 0 else None,
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/xconverter.py", line 131, in run
    xmodel = CORE.make_xmodel(model_files, model_type, _layout, in_shapes)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/core.py", line 104, in make_xmodel
    model_files, layout, in_shapes=in_shapes, model_type=model_t
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 97, in to_xmodel
    model_name, raw_nodes, layout, in_shapes, model_fmt, model_type
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 161, in create_xmodel
    xmodel = cls.__create_xmodel_from_tf1(name, layers, layout, in_shapes)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 267, in __create_xmodel_from_tf1
    xmodel_name, layout, layers, const_layer_dict, super_const_dict, in_shapes
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 3141, in __generate_xmodel
    f"*** Unsupported op: type: {op_type}, name: {op_name}"
ValueError: *** Unsupported op: type: scale, name: cnn/block1/bn0/FusedBatchNormV3/add
```

qianglin-xlnx, 2021-07-22:
Hi @FarhanAhmad4473 
Since this op is not supported by xcompiler, could you replace this op with another op, such as reshape? Thank you.

qianglin-xlnx, 2021-07-23:
Hi @FarhanAhmad4473 
There should be no scale op in tensorflow api. Could you please share your float model and quantized model, thus we can further analyze the problem. Thank you.

qianglin-xlnx, 2021-07-23:
Hi @FarhanAhmad4473 
I just note that you use ./quantize/deploy_model.pb as input.  Could you please use  ./quantize/quantize_eval_model.pb as input and try it again. Thank you.

qianglin-xlnx, 2022-03-03:
Hi @FarhanAhmad4473 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-03-03 by qianglin-xlnx

FarhanAhmad4473, 2022-03-03:
Yes, please close it. I have stopped working on it

zhenzhen-Xilinx, 2022-03-07:
This problem can be solved, please use the latest version.