Xilinx/Vitis-AI #466: Compilation aborted when compiling quantized model

andreaswestre, 2021-07-08:
Hello, 
I am trying to compile a custom pytorch model to run on a zcu104, by following [this tutorial](https://github.com/Xilinx/Vitis-Tutorials/tree/master/Machine_Learning/Design_Tutorials/09-mnist_pyt-master). I am using a simple custom model and have managed to quantize it, but are having problems compiling and running on the zcu104.
When running the final step 
`./compile.sh zcu104`
I get the following error:
`terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
Aborted (core dumped)`
Any ideas what the source of this empty array array can be? I have also posted this issue in the [Xilinx forums](https://forums.xilinx.com/t5/AI-and-Vitis-AI/Compile-a-custom-pytorch-model-to-run-on-a-zcu104/m-p/1261551).

qianglin-xlnx, 2021-07-22:
Hi @andreaswestre 
I wonder if your issue has been solved? If not, could you share your floating model, quantization script, dataset and quantized model? Thus we can reproduce the issue and do further analysis. Thank you.

MichaelX99, 2021-12-03:
I had this error and got around it by not using a bilinear upsample of 8 even though it is claimed to be supported in the docs, the most I could use was a 4x upsample. To achieve the 8x upsample I changed the 8x to a 4x upsample, followed by a 1x1 Conv2d+ReLU, followed by a 2x upsample

wanghong4compiler, 2022-02-19:
Hi @andreaswestre 
I wonder if your issue has been solved? If not, could you share your floating model, quantization script, dataset and quantized model? Thus we can reproduce the issue and do further analysis. Thank you.

wanghong4compiler, 2022-02-19:
Hi @MichaelX99 ,
Could you help share your original model that not split one upsample into two? Thus we can reproduce the issue and do further analysis, thank you.

qianglin-xlnx, 2022-03-03:
Hi @MichaelX99 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-03-03 by qianglin-xlnx