Xilinx/Vitis-AI #467: Question about depoly model: how to depoly a model that have multi inputs

TeresaKumo, 2021-07-09:
Hi,  I have quantized and compiled my pytorch model, but when I try to depoly my model on board(ZCU104), I found that I can't use dpu read two inputs, and I can't find any examples. 
Here is my model image use `xir svg ` ,  I have two inputs and concated them in my network.
[image]

Can you help me? Thanks a lot!

max271828, 2021-07-15:
You can use the interface of dpu_task to create such graph and you can set more than one input into your model. 
'dpu_task' is in vitis-ai-library, you can use its interface like following sample:

```
/*
 * Copyright 2019 Xilinx Inc.
 *
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 *     http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an "AS IS" BASIS,
 * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 */
#include <glog/logging.h>
#include <google/protobuf/text_format.h>
#include <vitis/ai/proto/dpu_model_param.pb.h>
#include <vitis/ai/library/tensor.hpp>

#include <cmath>
#include <iostream>
#include <numeric>
#include <opencv2/core.hpp>
#include <opencv2/highgui.hpp>
#include <opencv2/imgproc.hpp>
#include <vitis/ai/dpu_task.hpp>
#include <utility>

using namespace std;
using namespace cv;

int8_t dpu_round(float num) {
  if(num - floor(num) == 0.5) return ceil(num);
  else return round(num);
}

void copy_into_tensor (const vector<int8_t>& input,
                vitis::ai::library::InputTensor tensor) {
  int i_fixpos = tensor.fixpos;
  auto size = tensor.height * tensor.width * tensor.channel;
  CHECK_EQ(input.size(), tensor.size) << tensor.name << " right";
  for (size_t b = 0; b < tensor.batch; ++b) {
    auto data = (int8_t*)tensor.get_data(b);
    if(i_fixpos == 0) {
      memcpy(data, input.data() + b*size, size);
    } else {
      float scale = vitis::ai::library::tensor_scale(tensor);
      for(size_t i = 0; i < size; ++i)
        data[i] = dpu_round(input[i+b*size] * scale);
    }
    LOG(INFO)
      << "tensor shape: (" << tensor.height << ", " << tensor.width << ", " << tensor.channel << ").";
    if(1) {
      // it just for debug
      for(size_t i = 0; i < tensor.batch; ++i) {
        std::ofstream ofs("batch_"+tensor.name + to_string(i), ios::binary);
        ofs.write((char*)data, size);
        ofs.close();
      }
    }
  }
}
// reorder the tensors with name
vector<vitis::ai::library::InputTensor> sort_tensors (
            const vector<vitis::ai::library::InputTensor>& tensors, vector<string>& layer_names) {
  vector<vitis::ai::library::InputTensor> ordered_tensors;
  for (auto i = 0u; i < layer_names.size(); ++i)
    for (auto j = 0u; j < tensors.size(); ++j)
      if (tensors[j].name.find(layer_names[i]) != std::string::npos) {
        ordered_tensors.push_back(tensors[j]);
        break;
      }
  return ordered_tensors;
}

// run the model
cv::Mat inference(unique_ptr<vitis::ai::DpuTask>& task,
                  const cv::Mat& input_image) {

  auto input_unsort = task->getInputTensor(0u);
  vector<string> input_names = {"model_input_0", "model_input_1"};
  auto input_tensors = sort_tensors(input_unsort, input_names);

  auto sWidth = input_tensors[0].width;
  auto sHeight = input_tensors[0].height;

  if (0) {
  // if your input0 is just need to minus means and multiply scales, you can use this:
  // Set the mean values and scale values.
    task->setMeanScaleBGR({103.53, 116.28, 123.675}, {1.0, 1.0, 1.0});
    task->setImageRGB(input_image);
  }

  // copy your data into the input tennsors
  vector<int8_t> input0(input_tensors[0].size);
  vector<int8_t> input1(input_tensors[1].size);
  // do your preprocess for the two inputs

  copy_into_tensor(input0, input_tensors[0]);
  copy_into_tensor(input1, input_tensors[1]);

  task->run(0u);

  // do your post process
  auto output_tensor = task->getOutputTensor(0u)[0];

  auto output_data = (int8_t*)output_tensor.get_data(0);
  float f_scale = vitis::ai::library::tensor_scale(output_tensor);

  // you can use the output_data and f_scale to get your float output for your postprocess
  Mat ret; // = your final image.
  return ret;
}

int main(int argc, char* argv[]) {
  // A kernel name, it should be samed as the dnnc result. e.g.
  string kernel_name = string(argv[1]);
  Mat img = cv::imread(argv[2]);

  // Create a dpu task object.
  unique_ptr<vitis::ai::DpuTask> task = vitis::ai::DpuTask::create(kernel_name);

  auto result = inference(task, img);

  imshow("", result);
  waitKey(0);
  //imwrite("result.jpg", result[0]);
  return 0;
}
```
And you can compile it with this:
`g++ -std=c++17 -O3 -I. -o demo demo.cpp -lglog -lvitis_ai_library-model_config -lprotobuf -lvitis_ai_library-dpu_task -lopencv_core -lopencv_video -lopencv_videoio -lopencv_imgproc -lopencv_imgcodecs -lopencv_highgui`

Good Luck!

TeresaKumo, 2021-07-16:
> You can use the interface of dpu_task to create such graph and you can set more than one input into your model.
> 'dpu_task' is in vitis-ai-library, you can use its interface like following sample: [...]

Thanks for your clarifying,
Besides, can I use python to do the same things? (I use v2.6 image and have dpu overlay)

max271828, 2022-02-25:
Yes, it is almost sames as the c++, you can see the example under VART.

closed 2022-02-25 by qianglin-xlnx