Xilinx/Vitis-AI #468: April 21 Patch links not working

FarhanAhmad4473, 2021-07-09:
I am following 
https://github.com/Xilinx/Vitis-AI/blob/master/docs/quick-start/install/Vitis%20AI%201.3.2%20April%202021%20Patch.md
to download the patch and links are not live neither returns any error. 

Please advise?

qianglin-xlnx, 2021-07-15:
Hi @FarhanAhmad4473 
I can download the patches. Could you try to use another browser to download them?
Or you can use wget to download them, such as:
wget https://www.xilinx.com/bin/public/openDownload?filename=xnnc-1.3.2-py37_48.tar.bz2 -O xnnc-1.3.2-py37_48.tar.bz2

qianglin-xlnx, 2022-02-16:
Hi @FarhanAhmad4473 
Have you solved this issue? If you still can't download the patch, please let me know.
If it has been solved, please kindly close this issue. Thank you.

closed 2022-02-22 by qianglin-xlnx