Xilinx/Vitis-AI #471: Invalid Value and Overflows quantization vai-q-pytorch with .pt weights

Fabioni, 2021-07-19:
Hello,
I get the following output when I quantize a yolov4-csp model using a self trained .pt file for the weights.

[image]

When I convert the .pt file to darknets .weights file, and then use this everything works fine.
I use https://github.com/WongKinYiu/ScaledYOLOv4/tree/yolov4-csp

Can somebody see what goes wrong here?

Greetings

niuxjxlnx, 2021-07-22:
Hi, @Fabioni:

Python will do validatoin check of values, these warning of valid value and overflow is correct and useful to help clean up problems in .pt files. It is most likely some parameter values are unreasonable value becaused of not correctly initialized or loaded.

The converted darknet .weights file works fine because of some possible reasons:
1) unreasonable values has been handled to be reasonable during the conversion
2) C/C++ program has less reasonable value check than python

My suggestion is to print the values in .pt files and check why some values cause overflow.

Alexiazzf, 2022-03-09:
Hi @Fabioni ：

Thank you for your issue. We have added some protections for invalid value and overflows. This bug fix will be open sourced in the next version of vai-q-pytorch.

Best

closed 2022-03-09 by niuxjxlnx