Xilinx/Vitis-AI #472: Tensorflow2 quantization / compilation issue: input layer got quantized un-wantedly

luke-avionics, 2021-07-19:
I went through the quantization flow and got to compilation. However it complains some param_info was None (as shown below), I printed out more and found that Input layer is also quantized and its info is None
[image]
I am not sure quantizing the input layer is normal or not as it has no parameters. The following is part of my model definition 
[image]

luke-avionics, 2021-07-19:
something wrong with tqdm?

zhenzhen-Xilinx, 2022-03-07:
This problem can be solved, please use the latest version.

closed 2022-03-10 by qianglin-xlnx