Xilinx/Vitis-AI #474: 【WAA-TRD】pre-build for U200/U250

simplelins, 2021-07-22:
Hi, I need integration of pre/post processing accelerator with DPU fot the u200/u250 platform, where can I get the pre-project like U50 (https://github.com/Xilinx/Vitis-AI/tree/master/dsa/WAA-TRD/proj/pre-built/classification-pre_DPUv3e)?
or how to do some update for u200/u250?

sumitn-xilinx, 2021-07-22:
A sample U200/U250 WAA TRD would be available in the upcoming Vitis AI 1.4 release. Stay tuned.

sumitn-xilinx, 2021-07-26:
It's available here - https://github.com/Xilinx/Vitis-AI/tree/master/dsa/WAA-TRD/proj/build/classification-pre_DPUv3int8

simplelins, 2021-07-27:
> It's available here - https://github.com/Xilinx/Vitis-AI/tree/master/dsa/WAA-TRD/proj/build/classification-pre_DPUv3int8

https://www.xilinx.com/bin/public/openDownload?filename=xilinx-u200-gen3x16-xdma-all_1-3209015.deb.tar.gz   this link is unreach, please check at file 'install_u200_shell.sh'!

simplelins, 2021-07-27:
Application Exception: The XO file '../../../accel/DPUV3INT8/dpdpuv3_wrapper.xo' was created in version 2021.1, which is later than the software version you are currently running: 2020.2. Forward compatibility is not supported
Can you provide a lower version of the 'dpdpuv3_wrapper.xo'? because the version of  vivado is lower than 2021.1, thanks

sumitn-xilinx, 2021-07-27:
That would be difficult. The XO may be using the new features from 2021.1. Can you switch to 2021.1?

simplelins, 2021-07-27:
> That would be difficult. The XO may be using the new features from 2021.1. Can you switch to 2021.1?

ok， I will have a try!

simplelins, 2021-07-29:
Hi， I had updated the vivado to 2021.1, and installed the new shell xilinx_u250_gen3x16_xdma_shell_3_1. I had a test with vadd, it can run successfully. but when run the demo '/workspace/examples/DPUCADF8H/tf_resnet', the error as below :
```
(vitis-ai-tensorflow) Vitis-AI /workspace/examples/DPUCADF8H/tf_resnet > ./resnet_example ./xcompile/resnet50_v1_tf_compiled.xmodel ./images/
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0728 21:37:37.352671   137 main.cc:293] create running for subgraph: subgraph_resnet_v1_101/block1/unit_1/bottleneck_v1/add
terminate called after throwing an instance of 'std::runtime_error'
  what():  Error: Could not acquire CU
Aborted (core dumped)
```
we have the U200 and U250 in the machine. I also set the env `export XLNX_ENABLE_DEVICES=1`.
the stack is as below:
```
#0  0x00007fb7f92c5fb7 in raise () from /lib/x86_64-linux-gnu/libc.so.6
#1  0x00007fb7f92c7921 in abort () from /lib/x86_64-linux-gnu/libc.so.6
#2  0x00007fb7f9ab30a9 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007fb7f9abe506 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007fb7f9abe571 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007fb7f9abe7f5 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007fb7e2eea549 in XrmResource::XrmResource(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, xir::Attrs*) [clone .cold] ()
   from /usr/lib/librt-engine.so
#7  0x00007fb7e2f0146e in DeviceHandle::DeviceHandle(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, xir::Attrs*) ()
   from /usr/lib/librt-engine.so
#8  0x00007fb7e2f0210b in XrtDeviceHandle::XrtDeviceHandle(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, xir::Attrs*) ()
   from /usr/lib/librt-engine.so
#9  0x00007fb7e2f4675f in XclDpuController<XrtDeviceHandle, XrtDeviceBuffer, XrtDeviceBuffer>::XclDpuController(xir::Subgraph const*, xir::Attrs*) () from /usr/lib/librt-engine.so
#10 0x00007fb7e2f4f34b in Dpuv3Int8Controller::Dpuv3Int8Controller(xir::Subgraph const*) () from /usr/lib/librt-engine.so
#11 0x00007fb7e2ef90c7 in std::shared_ptr<DpuController> DpuControllerFactory::get<xir::Subgraph const*>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, xir::Subgraph const*, xir::Attrs*) () from /usr/lib/librt-engine.so
#12 0x00007fb7e2efb6ae in vart::DpuRunner::DpuRunner(xir::Subgraph const*) () from /usr/lib/librt-engine.so
#13 0x00007fb7e2efbc52 in create_runner () from /usr/lib/librt-engine.so
#14 0x00007fb7fb2ad55d in vart::Runner::create_runner(xir::Subgraph const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /usr/lib/libvart-runner.so.1
#15 0x0000558afeef7b09 in main (argc=3, argv=0x7ffef8053508)
    at /home/lww/project/12_FPGA/Vitis-AI/examples/DPUCADF8H/tf_resnet/src/main.cc:295
```

closed 2021-07-29 by simplelins