Xilinx/Vitis-AI #475: Build vai_c_xir

Uuuuuuho, 2021-07-22:
Hi, I'm working on a project of optimizing PointPillars model on ZCU102 board.
I've studied compilation flow including other architecture design of DPU core as well.
As far as I know, partial model of PointPillars (only PFE and RPN, accurately) are serialized (conversed) into *.xmodel format by script (run_quant.sh) in the demo project.

However, I can't find the source code of DPU instruction generator compiler (the Vitis-AI compiler).
I guess the consistently generated DPU instructions, regardless of the input models, are not optimized for all models because of computation bound or memory bound caused by data layout, data re-usability and so.

My question is as follows.
1. Is the Vitis-AI compiler not public so far?
2. If it is, is there any plan you will to make it so?
3. Is there any document explaining about DPU instruction set or compiler design?

qianglin-xlnx, 2021-07-22:
Hi @Uuuuuuho 
1. Vitis-AI compiler is not public.
2. So far as I know, there is no plan to open source Vitis-AI compiler.
3. I'm sorry. There is no document explaining about the DPU instruction set. For compiler, you can refer to https://www.xilinx.com/html_docs/vitis_ai/1_3/compiling_model.html#ztl1570696058091

Uuuuuuho, 2021-07-22:
Thanks for the quick reply.

I'd like to ask another question for memory management. 
I'm wondering how the input/output tensor and the intermediate value is managed.

How is the input/output tensor data transferred to the BRAM inside DPU core while executing assigned DPU kernels?

I guessed the data in the tensor buffer is managed by XRT driver, but it was not as far as I'm aware of.
In the Vitis-AI-Runtime library, the address buffer was sent to XRT in the _ert_ format.
However, I can't find the kernels related to the tensor data management in the XRT library as well.
The answer would be greatly helpful for me.

Thanks in advance.

closed 2021-07-22 by Uuuuuuho