Xilinx/Vitis-AI #481: still can't find packagegroup-petalinux-vitisai in vai 1.4

hill19850213, 2021-07-24:
Hi, 
I have referred the below step and do petalinux-upgrade, but I still can't find packagegroup-petalinux-vitisai

[image]

Can you share more detail?

qianglin-xlnx, 2021-07-26:
Hi @hill19850213 
So far as I know, the Petalinux tool updates containing VAI1.4 content is not released. It will be released 1 month later. Before that, you can install VART by the released packages under https://gitenterprise.xilinx.com/Vitis/vitis-ai-staging/tree/master/setup/mpsoc/VART/2021.1

hill19850213, 2021-07-26:
Hi @qianglin-xlnx ,
This link https://gitenterprise.xilinx.com/Vitis/vitis-ai-staging/tree/master/setup/mpsoc/VART/2021.1 is not available for me. Can you provide other link?

qianglin-xlnx, 2021-07-26:
@hill19850213 
Sorry about the mistake. It's https://github.com/Xilinx/Vitis-AI/tree/master/setup/mpsoc/VART/2021.1

hill19850213, 2021-07-30:
@qianglin-xlnx 
I try to use dnf to install packagegroup-petalinux-vitisai , but there is an error as below:
[image]
What is the problem?

Thanks
Sitting

qberthet, 2021-10-12:
@hill19850213 : I think you have to copy the rpm package to the target system and then install them locally with dnf (not tried yet).

@qianglin-xlnx : It seems that an update_1 is out ([http://petalinux.xilinx.com/sswreleases/rel-v2021/sdkupdate/2021.1_update1/aarch64](http://petalinux.xilinx.com/sswreleases/rel-v2021/sdkupdate/2021.1_update1/aarch64)), but it appear to be broken: I have updated my Petalinux installation with it but then when I try to configure a newly-created project, I get 

```
Extracting SDK......xz: (stdin): Compressed data is corrupt
tar: Child returned status 1
tar: Error is not recoverable: exiting now
ERROR: Failed to Extract Yocto SDK.
```
which seems to indicate that the binary blob in aarch64 is corrupt.
Do you have a clue? Or should I create a new issue?
Thanks in advance.

nhphuong91, 2021-11-26:
> @hill19850213 : I think you have to copy the rpm package to the target system and then install them locally with dnf (not tried yet).
>
> @qianglin-xlnx : It seems that an update_1 is out [...]

I'm having the same problem. Have you managed to fix it? @qberthet

qianglin-xlnx, 2021-11-26:
Hi @nhphuong91  and @hill19850213 
Could you try the following?
[image]

nhphuong91, 2021-11-26:
> Hi @nhphuong91 and @hill19850213 Could you try the following? [image]

@qianglin-xlnx it works! Thanks!

hill19850213, 2021-11-26:
@qianglin-xlnx it works. Thanks!

closed 2021-11-26 by hill19850213