Xilinx/Vitis-AI #484: Tensorflow 2 not supported by Ultra96v2

hghari, 2021-07-26:
Hi, I tried the example in vitis ai tutorial repository for signal modulation detection. I was on keras on tensorflow 2. The dpu does not give any results on ultra96. It outputs all zero values. Do you know the reason?
thanks

AVprog, 2021-07-27:
Perhaps there is a large current consumption of FPGA.

https://github.com/Xilinx/Vitis_Embedded_Platform_Source/blob/2019.2/Xilinx_Official_Platforms/zcu104_dpu/petalinux/project-spec/meta-user/recipes-ai/dpuclk/files/dpu_clk

This script will reduce the frequency of the PL_CLK0 (100MHz) clock source that feeds the Clock Wizard that generates the multiple clock frequencies, available for Vitis.

See: https://www.hackster.io/AlbertaBeef/vitis-ai-1-1-flow-for-avnet-vitis-platforms-part-1-007b0e

hoadv-qh, 2021-07-30:
Hi @hghari,

It is difficult to understand your situation without any sample of log or snip of code (your app code?).

>The dpu does not give any results on ultra96

You meant you interact with DPU correctly, but output data is zero after you run DPU?
If so, could you put the code which run DPU and the code which get output from DPU?

hghari, 2021-08-09:
Hi @AVprog  can the output being zero be related to clock rate?
Hi @hoadv-qh I am using the example in vitis tutorial. this link : https://github.com/Xilinx/Vitis-Tutorials/tree/master/Machine_Learning/Design_Tutorials/10-RF_modulation_recognition
I have trained it with both tensor 2 and tensor 1. Then quantized and compiled it successfully but when I use test_accuracy.py file it gives all zeros for the output. Thanks

hoadv-qh, 2021-08-12:
Hi @hghari
> I have trained it with both tensor 2 and tensor 1. Then quantized and compiled it successfully but when I use test_accuracy.py file it gives all zeros for the output. Thanks
You meant below, right?
* 1st: You created float model and trained, then got accuracy as your expectation
* 2nd: You quantized float model and got INT8 model. Then you evaluted INT8 model and also got accuracy as your expectation
* 3rd: You compiled INT8 model to Ultra96V2, and run on the board. BUT GOT ACCURACY 0%.

In this case, I prefer to debug on Ultra96V2 board with idea below.
* Print out outputData[j] whether output tesors has "normal" data.
* You can try to do post processing to get human-readable output by using same way as Float model or INT8 model
(as in vai_2018_RadioML_VAI_keras.ipynb)
I saw that post process of those models only uses numpy. So it is easy to run on Ultra96V2.