Xilinx/Vitis-AI #487: 使用Vitis AI量化训练Pytorch模型出现问题

zhangchenchen-novauto, 2021-07-28:
看报错！但所我已经改了里面含有Add节点的地方
[image]

第335行，
```
# x = x + out[module.layers[0]]
 x = self.skip_add[i](x, out[module.layers[0]])
```
[models.txt](https://github.com/Xilinx/Vitis-AI/files/6893139/models.txt)

执行到这句代码就出现了上面的错误
quantizer = torch_quantizer(quant_mode='calib',
                              module=model,
                              input_args=input,
                              bitwidth=8,
                              qat_proc=True)

yuwang-xlnx, 2021-07-29:
Hi @zhangchenchen-novauto,

Which Vitis-AI verison did you use ? Can you share the quant_info.json and python script in .qat ?

zhangchenchen-novauto, 2021-07-29:
**Vitis-AI1.4** 
这是.qat/ 文件夹下面的文件

[quant_info.json.txt](https://github.com/Xilinx/Vitis-AI/files/6897804/quant_info.json.txt)
[Darknet.py.txt](https://github.com/Xilinx/Vitis-AI/files/6897807/Darknet.py.txt)

yuwang-xlnx, 2021-07-29:
> Hi @yuwang-xlnx
> **Vitis-AI1.4**
> 这是.qat/ 文件夹下面的文件
> 
> [quant_info.json.txt](https://github.com/Xilinx/Vitis-AI/files/6897804/quant_info.json.txt)
> [Darknet.py.txt](https://github.com/Xilinx/Vitis-AI/files/6897807/Darknet.py.txt)

It seems that the replacement of torch.add/+ is not correct. 
A simple example:

```
from pytorch_nndct.nn.modules import functional

def __init__(self):
  self.add = functional.Add()
  ...

def forward(self, x):
  ...
  x = self.add(x, y)
```

zhangchenchen-novauto, 2021-07-29:
Hi @yuwang-xlnx 
我感觉我的替换好像没有问题，在问题提问的 **model.txt**   https://github.com/Xilinx/Vitis-AI/files/6893139/models.txt 有我的网络描述
[image]
[image]

yuwang-xlnx, 2021-09-17:
Hi @zhangchenchen-novauto ,

There should be xxx.py and quant_info.json in the .qat directory, can you share these two files ?

closed 2022-02-25 by niuxjxlnx