Xilinx/Vitis-AI #488: Custom model has four outputs, VAI Compiler generates single output.

vtorres621, 2021-07-28:
Hello, 
I am running a custom model through the Vitis AI flow.  I need to postprocess the output of 3 intermediate layers inside the model as well as the final output .  My pytorch model reflects this.  The model's forward function generated by VAI Quantizer also returns 4  required outputs:
`return self.output_module_18,self.output_module_34,self.output_module_50,self.output_module_66`

But when I compile the subgraphs, the .xmodel has only the output of the final layer.  

[image]

I have seen other models from the model zoo (like yolov3, tinyyolo, etc) "download" the output of an intermediate layer:
[image]

 How can I do this with my custom layer?

 
Let me know if you need more information.

hoadv-qh, 2021-07-30:
Hi @vtorres621,

>But I am not sure where to fin the "op_name0".

In your graph, you can see such "Tensor: CNN__CNN_Sequential_xxx".
I think you can try to pass name of tensor, "CNN__CNN_Sequential_xxx", to the option?

vtorres621, 2021-07-30:
Hello @hoadv-qh,

Thanks for your help, I was trying this with Vitis AI 1.3 but it didn't worked.  
I updated to 1.4 and I was able to get multiple outputs using this --options command on VAI_C/

Thanks!

closed 2021-07-30 by vtorres621

hoadv-qh, 2021-07-30:
Hi @vtorres621,

I'm glad that can help.
Hope you can do well for next steps with DPU.