Xilinx/Vitis-AI #489: [bug report] libvart-runner cause core dumped, at the version v1.4

simplelins, 2021-07-29:
Hi,
  I try to build the ` Vitis-AI-Runtime` debug libs, the branch tag is v1.4, the lib `libvart-runner.so`  cause a 'core dumped ',
  ```
  (vitis-ai-tensorflow) lww@FPGAServer:/home/lww/project/12_FPGA/Vitis-AI/examples/DPUCADF8H/tf_resnet$ ./resnet_example ./xcompile/resnet50_v1_tf_compiled.xmodel ./images/
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0729 17:13:11.894138 10577 main.cc:293] create running for subgraph: subgraph_resnet_v1_101/block1/unit_1/bottleneck_v1/add
Segmentation fault (core dumped)
  ``` 
   stack is as below:
  ```

  Thread 1 "resnet_example" received signal SIGSEGV, Segmentation fault.
0x00007fffdd320184 in Xmodel::getInH() () from /opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/librt-engine.so
(gdb) bt
#0  0x00007fffdd320184 in Xmodel::getInH() () from /opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/librt-engine.so
#1  0x00007fffdd30be73 in Dpuv3Int8Controller::initializeTensors() ()
   from /opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/librt-engine.so
#2  0x00007fffdd30fa91 in Dpuv3Int8Controller::Dpuv3Int8Controller(xir::Subgraph const*) ()
   from /opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/librt-engine.so
#3  0x00007fffdd2b91b7 in std::shared_ptr<DpuController> DpuControllerFactory::get<xir::Subgraph const*>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, xir::Subgraph const*, xir::Attrs*) ()
   from /opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/librt-engine.so
#4  0x00007fffdd2bb79e in vart::DpuRunner::DpuRunner(xir::Subgraph const*) ()
   from /opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/librt-engine.so
#5  0x00007fffdd2bbd42 in create_runner () from /opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/librt-engine.so
#6  0x00007ffff7b86087 in vart::Runner::create_runner (subgraph=0x555575a6ebd0, mode=...)
    at /Data/lww/project/12_FPGA/Vitis-AI/tools/Vitis-AI-Runtime/VART/vart/runner/src/dpu_runner.cpp:165
#7  0x000055555555d46f in main ()
(gdb) q
A debugging session is active.

  ```
  please cheack!

wilderfield, 2021-07-29:
> (vitis-ai-tensorflow) ***@***.***:/home/lww/project/12_FPGA/Vitis-AI/examples/DPUCADF8H/tf_resnet$ ./resnet_example ./xcompile/resnet50_v1_tf_compiled.xmodel ./images/

Did you compile the model yourself? Or is it precompiled from the model zoo?

simplelins, 2021-07-30:
Yeah, I recompiled the model. I can run the model with docker env successfully, but when using the lib recompiling by myself that cause the core dump. and alse when I remove the recompiling lib 'libvart-runner',It run successfully again.

simplelins, 2021-08-06:
@wilderfield   please check!    how to get a debug vart libs that can run successfully at Vitis-AI v1.4?

closed 2021-08-23 by simplelins