Xilinx/Vitis-AI #490: vai_c_caffe resnet50 on U250 error!

zy486at189cn, 2021-07-29:
use:
vai_c_caffe  --prototxt results/deploy.prototxt --caffemodel results/deploy.caffemodel --output_dir results/ --net_name ResNet_50 --arch /opt/vitis_ai/compiler/arch/DPUCADF8H/U250/arch.json --options    "{'mode':'normal', 'save_kernel':''}"

result:
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(batchsize=1, inputs_shape=None, layout='NCHW', model_files=['results/deploy.caffemodel'], model_type='caffe', named_inputs_shape=None, out_filename='/tmp/ResNet_50_org.xmodel', proto='results/deploy.prototxt')
[INFO] caffe model: /workspace/program/SAR/dpu/results/deploy.caffemodel
[INFO] caffe model: /workspace/program/SAR/dpu/results/deploy.prototxt
[INFO] parse raw model     :100%|█| 194/194 [00:08<00:00, 23.11it/s]            
[INFO] infer shape (NCHW)  :100%|█| 194/194 [00:00<00:00, 1692.56it/s]          
[INFO] infer shape (NHWC)  :100%|█| 194/194 [00:00<00:00, 3534.95it/s]          
[INFO] perform level-1 opt :100%|█| 3/3 [00:00<00:00, 170.72it/s]               
[INFO] infer shape (NHWC)  :100%|█| 196/196 [00:00<00:00, 4581.69it/s]          
[INFO] generate xmodel     :100%|█| 196/196 [00:00<00:00, 737.84it/s]           
[INFO] dump xmodel: /tmp/ResNet_50_org.xmodel
[UNILOG][INFO] Target architecture: DPUCADF8H_ISA0
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCADF8H_ISA0
[UNILOG][INFO] Graph name: deploy, with op num: 412
[UNILOG][INFO] Begin to compile...
[UNILOG][WARNING] DPU prefers xir::Op{name = conv1, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[UNILOG][WARNING] xir::Op{name = conv1, type = conv2d-fix} has been assigned to CPU.
[UNILOG][WARNING] DPU prefers xir::Op{name = pool1, type = pool-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[UNILOG][WARNING] xir::Op{name = pool1, type = pool-fix} has been assigned to CPU.
[UNILOG][WARNING] DPU prefers xir::Op{name = res2a_branch1, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[UNILOG][WARNING] xir::Op{name = res2a_branch1, type = conv2d-fix} has been assigned to CPU.
[UNILOG][WARNING] DPU prefers xir::Op{name = res2a_branch2a, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[UNILOG][WARNING] xir::Op{name = res2a_branch2a, type = conv2d-fix} has been assigned to CPU.
[UNILOG][WARNING] DPU prefers xir::Op{name = res2a_branch2b, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[UNILOG][WARNING] xir::Op{name = res2a_branch2b, type = conv2d-fix} has been assigned to CPU.
[UNILOG][WARNING] DPU prefers xir::Op{name = res2a_branch2c, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[UNILOG][WARNING] xir::Op{name = res2a_branch2c, type = conv2d-fix} has been assigned to CPU.
[UNILOG][WARNING] DPU prefers xir::Op{name = res2a, type = eltwise-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[UNILOG][WARNING] xir::Op{name = res2a, type = eltwise-fix} has been assigned to CPU.
[UNILOG][WARNING] DPU prefers xir::Op{name = res2b_branch2a, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[UNILOG][WARNING] xir::Op{name = res2b_branch2a, type = conv2d-fix} has been assigned to CPU.
[UNILOG][WARNING] DPU prefers xir::Op{name = res2b_branch2b, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[UNILOG][WARNING] xir::Op{name = res2b_branch2b, type = conv2d-fix} has been assigned to CPU.
[UNILOG][WARNING] DPU prefers xir::Op{name = res2b_branch2c, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[UNILOG][WARNING] xir::Op{name = res2b_branch2c, type = conv2d-fix} has been assigned to CPU.
[UNILOG][WARNING] DPU prefers xir::Op{name = res2b, type = eltwise-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[UNILOG][WARNING] xir::Op{name = res2b, type = eltwise-fix} has been assigned to CPU.
[UNILOG][WARNING] DPU prefers xir::Op{name = res2c_branch2a, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[UNILOG][WARNING] xir::Op{name = res2c_branch2a, type = conv2d-fix} has been assigned to CPU.
[UNILOG][WARNING] DPU prefers xir::Op{name = res2c_branch2b, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[UNILOG][WARNING] xir::Op{name = res2c_branch2b, type = conv2d-fix} has been assigned to CPU.
[UNILOG][WARNING] DPU prefers xir::Op{name = res2c_branch2c, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[UNILOG][WARNING] xir::Op{name = res2c_branch2c, type = conv2d-fix} has been assigned to CPU.
[UNILOG][WARNING] DPU prefers xir::Op{name = res2c, type = eltwise-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[UNILOG][WARNING] xir::Op{name = res2c, type = eltwise-fix} has been assigned to CPU.
[UNILOG][WARNING] DPU prefers xir::Op{name = fake_downsample_5, type = downsample-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[UNILOG][WARNING] xir::Op{name = fake_downsample_5, type = downsample-fix} has been assigned to CPU.
[UNILOG][WARNING] DPU prefers xir::Op{name = res3a_branch1, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[UNILOG][WARNING] xir::Op{name = res3a_branch1, type = conv2d-fix} has been assigned to CPU.
[UNILOG][WARNING] DPU prefers xir::Op{name = fake_downsample_0, type = downsample-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[UNILOG][WARNING] xir::Op{name = fake_downsample_0, type = downsample-fix} has been assigned to CPU.
[UNILOG][WARNING] DPU prefers xir::Op{name = res3a_branch2a, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[UNILOG][WARNING] xir::Op{name = res3a_branch2a, type = conv2d-fix} has been assigned to CPU.
[UNILOG][WARNING] DPU prefers xir::Op{name = res3a_branch2b, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[UNILOG][WARNING] xir::Op{name = res3a_branch2b, type = conv2d-fix} has been assigned to CPU.
[UNILOG][WARNING] DPU prefers xir::Op{name = res3a_branch2c, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[UNILOG][WARNING] xir::Op{name = res3a_branch2c, type = conv2d-fix} has been assigned to CPU.
[UNILOG][WARNING] DPU prefers xir::Op{name = res3a, type = eltwise-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[UNILOG][WARNING] xir::Op{name = res3a, type = eltwise-fix} has been assigned to CPU.
[UNILOG][WARNING] DPU prefers xir::Op{name = res3b_branch2a, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[UNILOG][WARNING] xir::Op{name = res3b_branch2a, type = conv2d-fix} has been assigned to CPU.
[UNILOG][WARNING] DPU prefers xir::Op{name = res3b_branch2b, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[... 122 lines cut ...]
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(batchsize=1, inputs_shape=None, layout='NCHW', model_files=['results/deploy.caffemodel'], model_type='caffe', named_inputs_shape=None, out_filename='/tmp/ResNet_50_org.xmodel', proto='results/deploy.prototxt')
[INFO] caffe model: /workspace/program/SAR/dpu/results/deploy.caffemodel
[INFO] caffe model: /workspace/program/SAR/dpu/results/deploy.prototxt
[INFO] parse raw model     :100%|█| 194/194 [00:08<00:00, 23.39it/s]            
[INFO] infer shape (NCHW)  :100%|█| 194/194 [00:00<00:00, 1684.82it/s]          
[INFO] infer shape (NHWC)  :100%|█| 194/194 [00:00<00:00, 3545.60it/s]          
[INFO] perform level-1 opt :100%|█| 3/3 [00:00<00:00, 171.60it/s]               
[INFO] infer shape (NHWC)  :100%|█| 196/196 [00:00<00:00, 4633.08it/s]          
[INFO] generate xmodel     :100%|█| 196/196 [00:00<00:00, 774.57it/s]           
[INFO] dump xmodel: /tmp/ResNet_50_org.xmodel
[UNILOG][INFO] Target architecture: DPUCAHX8H_ISA2
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCAHX8H_ISA2
[UNILOG][INFO] Graph name: deploy, with op num: 412
[UNILOG][INFO] Begin to compile...
[UNILOG][INFO] Total device subgraph number 3, DPU subgraph number 1
[UNILOG][INFO] Compile done.
[UNILOG][INFO] The meta json is saved to "/workspace/program/SAR/dpu/results/meta.json"
[UNILOG][INFO] The compiled xmodel is saved to "/workspace/program/SAR/dpu/results//ResNet_50.xmodel"
[UNILOG][INFO] The compiled xmodel's md5sum is ca883363b5cf86828671d5047dc70ee5, and has been saved to "/workspace/program/SAR/dpu/results/md5sum.txt"

the xmodel can be used under VART.

woinck, 2021-07-30:
U250 only support batchsize 4. Could you try adding the `input_shape` key in vai_c_caffe's `--options`  argument? For example:
```
--options "{'mode':'normal', 'save_kernel':'', 'input_shape': '4,3,224,224'}"
```
and replace the input channel/height/width with yours

zy486at189cn, 2021-07-30:
@woinck nice job!