Xilinx/Vitis-AI #491: Error at calibration when quantize PyTorch LSTM model

hoadv-qh, 2021-07-30:
Hi developers,

I'm quantizing a model with PyTorch's LSTM (torch.nn.LSTM). However, I got error at calibration step for model as below.
I feel that is is nearly final step of calibration, that is Quantizer generating something.

I used vitis-ai-cpu:1.3.411 to quantize model. Looks that Quantizer is looking for "name" attribute from LSTM.
Could you hint me how to come over this issues to continue quantizing model?

Thank you very much!

Error log.
<pre>
[NNDCT_NOTE]: =>Quantizable module is generated.(quant_model/Model.py)
Traceback (most recent call last):
  File "quantize.py", line 181, in <module>
    quantize(opt)
  File "quantize.py", line 70, in quantize
    quantizer = torch_quantizer(opt.quant_mode, float_model, (rand_in, text_for_pred), output_dir=opt.quant_model)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/apis.py", line 71, in __init__
    lstm_app = lstm_app)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py", line 119, in __init__
    device=device)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py", line 166, in prepare_quantizable_module
    connect_module_with_graph(quant_module, graph)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py", line 107, in connect_module_with_graph
    ModuleHooker.hook_module_with_node(module, graph)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/ModuleHooker.py", line 174, in hook_module_with_node
    cls.apply_to_children(module, _add_node_on_module)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/ModuleHooker.py", line 69, in apply_to_children
    child.apply(func)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/torch/nn/modules/module.py", line 287, in apply
    fn(self)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/ModuleHooker.py", line 165, in _add_node_on_module
    op.params_name = [v.name for v in node.op.params.values()]
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/ModuleHooker.py", line 165, in <listcomp>
    op.params_name = [v.name for v in node.op.params.values()]
AttributeError: 'list' object has no attribute 'name'
</pre>

hoadv-qh, 2021-07-31:
To be noted that I plan to deploy the above model to Ultra96V2 board.

qianglin-xlnx, 2021-09-17:
Hi @hoadv-qh,
Please note that for LSTM, currently only U50lv and U25 boards are supported.

hoadv-qh, 2021-09-18:
Hi @qianglin-xlnx,

Thank you very much for your answer.
However, I have several concerns just to confirm my understanding.

* (1) LSTM only for U50lv and U25, because only their DPUs are suitable for recurrent network or just limited by Vitis-AI version?
* (2) If it limits by Vitis-AI version, then do we have plan when Vitis-AI supports LSTM for such DPU used in Ultra96V2 board?
* (3) If it due to DPU HW, then is it possible to reconfigure/recreate DPU for Ultra96V2 by HW tool (Like Vivado) to support LSTM?
(it helps a lot when there are reference of DPU configure for custom purpose) 

Thank you for your help!

qianglin-xlnx, 2021-09-18:
Hi @hoadv-qh 
It's due to the DPU IP. And so far as I know, there is no plan to develop RNN DPU for Mpsoc platform, such as Ultra96V2.

hoadv-qh, 2021-09-18:
Hi @qianglin-xlnx,

Thank you for your information.
I understood.

hoadv-qh, 2021-09-18:
I closed this issued since it was clear that due to DPU IP.
(also, I updated issue subject to mention clearly target board for other if they have same question in future)

closed 2021-09-18 by hoadv-qh