Xilinx/Vitis-AI #492: ERROR at WAA-TRD Pre-processing Accelerator .xo Compiling

HSqure, 2021-07-30:
### Environment
`Vitis-Ai v1.4`
`Vitis suit 2021.1`
`XRT 2021.1`
`zcu102 platform file 2021.1`

### Operation
Running according to [WAA detection demo Readme document](https://github.com/Xilinx/Vitis-AI/tree/master/dsa/WAA-TRD/proj/build/detection-pre_DPUv2). ERROR occurred at pre-processing `.xo` kernel compiling step (HLS).

### ERROR Log
```
----
Compiling kernel pp_pipeline_accel...
mkdir -p /home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre
v++ -o /home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/pp_pipeline_accel.xo --kernel pp_pipeline_accel --compile /home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/xf_pp_pipeline_accel.cpp \
	-I. -I/home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/include -D__SDSVHLS__ -DHLS_NO_XIL_FPO_LIB --clock.defaultFreqHz 300000000 -I/home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre --target hw --platform /home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/platform/xilinx_zcu102_base_202110_1/xilinx_zcu102_base_202110_1.xpfm --temp_dir /home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/_x_xilinx_zcu102_base_202110_1_hw --save-temps --debug --kernel_frequency 300 --report_level 2 --xp vivado_prop:run.impl_1.strategy=Performance_Explore
WARNING: [v++ 60-1600] The option 'xp' was used directly on the command line, where its usage is deprecated. To ensure input line works for supported operating systems or shells, v++ supports specification for some options in a configuration file. As an alternative, please use options 'advanced.*', 'vivado.*' in a configuration file. Use one or more configuration files along with section headers to define key-value pairs for the advanced properties or parameters. Specify a configuration file using '--config'.
INFO: [v++ 82-185] Check out the auto-generated 'sample_compile.ini' configuration file. The file shows how to migrate from deprecated command line --xp switches to configuration file directives.
Option Map File Used: '/opt/Xilinx/Vitis/2021.1/data/vitis/vpp/optMap.xml'

****** v++ v2021.1 (64-bit)
  **** SW Build 3246112 on 2021-06-09-14:19:56
    ** Copyright 1986-2020 Xilinx, Inc. All Rights Reserved.

INFO: [v++ 60-1306] Additional information associated with this v++ compile can be found at:
	Reports: /home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/_x_xilinx_zcu102_base_202110_1_hw/reports/pp_pipeline_accel
	Log files: /home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/_x_xilinx_zcu102_base_202110_1_hw/logs/pp_pipeline_accel
Running Dispatch Server on port: 42381
INFO: [v++ 60-1548] Creating build summary session with primary output /home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/pp_pipeline_accel.xo.compile_summary, at Fri Jul 30 16:47:04 2021
INFO: [v++ 60-1316] Initiating connection to rulecheck server, at Fri Jul 30 16:47:04 2021
Running Rule Check Server on port:37767
INFO: [v++ 60-1315] Creating rulecheck session with output '/home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/_x_xilinx_zcu102_base_202110_1_hw/reports/pp_pipeline_accel/v++_compile_pp_pipeline_accel_guidance.html', at Fri Jul 30 16:47:05 2021
INFO: [v++ 60-895]   Target platform: /home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/platform/xilinx_zcu102_base_202110_1/xilinx_zcu102_base_202110_1.xpfm
INFO: [v++ 60-1578]   This platform contains Xilinx Shell Archive '/home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/platform/xilinx_zcu102_base_202110_1/hw/xilinx_zcu102_base_202110_1.xsa'
INFO: [v++ 60-585] Compiling for hardware target
INFO: [v++ 60-423]   Target device: xilinx_zcu102_base_202110_1
INFO: [v++ 60-242] Creating kernel: 'pp_pipeline_accel'
INFO: [v++ 60-1616] Creating a HLS clock using kernel_frequency option: 300 MHz

===>The following messages were generated while  performing high-level synthesis for kernel: pp_pipeline_accel Log file: /home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/_x_xilinx_zcu102_base_202110_1_hw/pp_pipeline_accel/pp_pipeline_accel/vitis_hls.log :
ERROR: [v++ 207-3772] use of undeclared identifier 'iTC'; did you mean 'TC'?: /home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/include/imgproc/xf_cvt_color.hpp:6047:32
ERROR: [v++ 207-3772] use of undeclared identifier 'iTC'; did you mean 'TC'?: /home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/include/imgproc/xf_cvt_color.hpp:6047:40
ERROR: [v++ 207-3772] use of undeclared identifier 'iTC'; did you mean 'TC'?: /home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/include/imgproc/xf_cvt_color.hpp:6151:32
ERROR: [v++ 207-3772] use of undeclared identifier 'iTC'; did you mean 'TC'?: /home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/include/imgproc/xf_cvt_color.hpp:6151:40
ERROR: [v++ 207-3772] use of undeclared identifier 'iTC'; did you mean 'TC'?: /home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/include/imgproc/xf_cvt_color.hpp:6231:32
ERROR: [v++ 207-3772] use of undeclared identifier 'iTC'; did you mean 'TC'?: /home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/include/imgproc/xf_cvt_color.hpp:6231:40
ERROR: [v++ 207-3772] use of undeclared identifier 'iTC'; did you mean 'TC'?: /home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/include/imgproc/xf_cvt_color.hpp:6744:32
ERROR: [v++ 207-3772] use of undeclared identifier 'iTC'; did you mean 'TC'?: /home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/include/imgproc/xf_cvt_color.hpp:6744:40
ERROR: [v++ 207-3772] use of undeclared identifier 'iTC'; did you mean 'TC'?: /home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/include/imgproc/xf_cvt_color.hpp:6849:32
ERROR: [v++ 207-3772] use of undeclared identifier 'iTC'; did you mean 'TC'?: /home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/include/imgproc/xf_cvt_color.hpp:6849:40
ERROR: [v++ 207-3772] use of undeclared identifier 'iTC'; did you mean 'TC'?: /home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/include/imgproc/xf_cvt_color.hpp:6928:32
ERROR: [v++ 207-3772] use of undeclared identifier 'iTC'; did you mean 'TC'?: /home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/include/imgproc/xf_cvt_color.hpp:6928:40
ERROR: [v++ 60-300] Failed to build kernel(ip) pp_pipeline_accel, see log for details: /home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/_x_xilinx_zcu102_base_202110_1_hw/pp_pipeline_accel/pp_pipeline_accel/vitis_hls.log
ERROR: [v++ 60-773] In '/home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/_x_xilinx_zcu102_base_202110_1_hw/pp_pipeline_accel/pp_pipeline_accel/vitis_hls.log', caught Tcl error: ERROR: [HLS 207-3772] use of undeclared identifier 'iTC'; did you mean 'TC'?: /home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/include/imgproc/xf_cvt_color.hpp:6047:32
ERROR: [v++ 60-773] In '/home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/_x_xilinx_zcu102_base_202110_1_hw/pp_pipeline_accel/pp_pipeline_accel/vitis_hls.log', caught Tcl error: ERROR: [HLS 207-3772] use of undeclared identifier 'iTC'; did you mean 'TC'?: /home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/include/imgproc/xf_cvt_color.hpp:6047:40
ERROR: [v++ 60-773] In '/home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/_x_xilinx_zcu102_base_202110_1_hw/pp_pipeline_accel/pp_pipeline_accel/vitis_hls.log', caught Tcl error: ERROR: [HLS 207-3772] use of undeclared identifier 'iTC'; did you mean 'TC'?: /home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/include/imgproc/xf_cvt_color.hpp:6151:32
ERROR: [v++ 60-773] In '/home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/_x_xilinx_zcu102_base_202110_1_hw/pp_pipeline_accel/pp_pipeline_accel/vitis_hls.log', caught Tcl error: ERROR: [HLS 207-3772] use of undeclared identifier 'iTC'; did you mean 'TC'?: /home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA [... cut ...]
Makefile:299: recipe for target '/home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/pp_pipeline_accel.xo' failed
make: *** [/home/alinx/huahua/workspace/Vitis-AI-master-v1.4/dsa/WAA-TRD/accel/detection-pre/pp_pipeline_accel.xo] Error 1

```

It's strange that the HLS source code is totally the same as `vitis-ai v1.3` version, it's smooth when I compiling it via vitis 2020.2. But unfortunately, it compiles failed when I try to do the same thing according to the document at `vitis-ai v1.4` version.

Thank you very much If there are any suggestions.

sumitn-xilinx, 2021-08-02:
This app is compatible with Vitis 2020.2 only. You can build the xclbin with Vitis 2020.2 and use it with Vitis AI 1.4.

HSqure, 2021-08-04:
> This app is compatible with Vitis 2020.2 only. You can build the xclbin with Vitis 2020.2 and use it with Vitis AI 1.4.

Thanks! I try to replace the `include` folder with the latest Vitis library from another repository and it worked.

closed 2021-08-04 by HSqure