Xilinx/Vitis-AI #493: VAI_C_TENSORFLOW cannot compiled 16-bit fixed-point

hoadv-qh, 2021-07-31:
Hi,

I want to increase precision for my YOLO model, then I quantized model with 2 options below to use 16-bit fixed-point instead of 8-bit fixed-point (default).
<pre>
vai_q_tensorflow quantize ...
                          --weight_bit 16 \
                          --activation_bit 16 \
                          ...
</pre>

However, when compiled model, I got errors from *vai_c_tensorflow*, and couldn't compiling successfully.
I had tried with 2 versions of  **vitis-ai-cpu**, **1.3.411** and **1.4.916**.

I'd like to know whether it dues to problem of compling or limitation of target DPU?
* If compiling problems, please suggest the way to use more than 8-bit for fixed-point
* If due to target DPU limitation, I think we had better to make error log become more precisely/understandable

Error log from vitis-ai-cpu:1.3.411
<pre>
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(inputs_shape=['1,416,416,3'], layout='NHWC', model_files=['../02_quantize/yolov4_quantized/quantize_eval_model.pb'], model_type='tensorflow', out_filename='yolov4_compiled//dpu_yolov4_tiny_Ultra96v2_org.xmodel', proto=None)
in_shapes: [[1, 416, 416, 3]]
[INFO] tensorflow model: ../02_quantize/yolov4_quantized/quantize_eval_model.pb
[INFO] parse raw model     :100%|███████████████████████████████████████████████████| 108/108 [00:00<00:00, 23262.20it/s]
[INFO] infer shape (NHWC)  :100%|███████████████████████████████████████████████████| 90/90 [00:00<00:00, 2625.89it/s]
[INFO] infer shape (NHWC)  :100%|███████████████████████████████████████████████████| 90/90 [00:00<00:00, 3927.54it/s]
[INFO] generate xmodel     :100%|███████████████████████████████████████████████████| 90/90 [00:00<00:00, 2325.00it/s]
[INFO] generate xmodel: /workspace/03_compiled/yolov4_compiled/dpu_yolov4_tiny_Ultra96v2_org.xmodel
[UNILOG][INFO] The compiler log will be dumped at "/tmp/vitis-ai-user/log/xcompiler-20210730-205816-123"
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA0_B2304_MAX_BG2
[UNILOG][INFO] Graph name: quantize_eval_model, with op num: 174
[UNILOG][INFO] Begin to compile...
[UNILOG][Check Failed: 0][/home/xbuild/conda-bld/xcompiler_1606385139459/work/src/util/CommonTypes.cpp:362][XCOM_ITEM_UNDEFINED][The requested item is not finded or defined!]
*** Check failure stack trace: ***
Aborted (core dumped)
</pre> 

Error log from vitis-ai-cpu:1.4.916.
<pre>
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(batchsize=1, inputs_shape=['1,416,416,3'], layout='NHWC', model_files=['../02_quantize/yolov4_quantized/quantize_eval_model.pb'], model_type='tensorflow', named_inputs_shape=None, out_filename='/tmp/dpu_yolov4_tiny_Ultra96v2_org.xmodel', proto=None)
in_shapes: [[1, 416, 416, 3]]
[INFO] tensorflow model: /workspace/02_quantize/yolov4_quantized/quantize_eval_model.pb
[INFO] parse raw model     :100%|███████████████████████████████████████████████████| 108/108 [00:00<00:00, 18253.74it/s]
[INFO] infer shape (NHWC)  :100%|███████████████████████████████████████████████████| 90/90 [00:00<00:00, 1101.14it/s]
[INFO] perform level-0 opt :100%|███████████████████████████████████████████████████| 2/2 [00:00<00:00, 544.33it/s]
[INFO] perform level-1 opt :100%|███████████████████████████████████████████████████| 6/6 [00:00<00:00, 685.49it/s]
[INFO] generate xmodel     :100%|███████████████████████████████████████████████████| 90/90 [00:00<00:00, 1897.22it/s]
[INFO] dump xmodel: /tmp/dpu_yolov4_tiny_Ultra96v2_org.xmodel
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA0_B2304_MAX_BG2
[UNILOG][INFO] Graph name: quantize_eval_model, with op num: 174
[UNILOG][INFO] Begin to compile...
[UNILOG][FATAL][XCOM_ITEM_UNDEFINED][The requested item is not finded or defined!]
*** Check failure stack trace: ***
Aborted (core dumped)
</pre>

hoadv-qh, 2021-07-31:
After looking into [DPU Naming](https://github.com/Xilinx/Vitis-AI/blob/master/docs/learn/dpu_naming.md), I think the compilation errors relate to specification of DPU.
According to above log, target DPU is **DPUCZDX8G**,  the number "8" means that target DPU only supports 8-bit quantization.

If so, I think error message should be updated to show why/solutions for user, instead of something is not defined in above log.

vtorres621, 2021-09-24:
Hello, did you find another 16-bit DPU alternative? I need a higher bit-depth for a particular project but have not been able to find one for the ZCU102 board.

joGu14, 2021-11-04:
Hello @hoadv-qh and @vtorres621. Did you find any solution? I'm stuck with the same problem. Thanks in advance.

vtorres621, 2021-11-04:
I could not find a solution.
Ended up using the 8-bit DPU.

hoadv-qh, 2021-11-07:
Hi @joGu14

As same as @vtorres621, I also ended up using 8-bit of quantization.
As I mentioned before, looks that it depends on DPU architecture (=HW characteristics itself).
However, I don't know whether any update from Xilinx after 3 months (from my first post) or not.

joGu14, 2021-11-10:
Thanks for your rapid responses.

wanghong4compiler, 2022-02-16:
Hi @hoadv-qh ,
for current hardware design, DPU part only supports 8-bit quantized models.

hoadv-qh, 2022-02-18:
Hi all,
Thank you for your discussion and information.
I think we can close this topic here.

I closed this topic!