Xilinx/Vitis-AI #494: Failed to have standalone BatchNorm2d implemented on DPU for both Vitis-AI 1.3 and 1.4

FrankXinqi, 2021-07-31:
I am using PyTorch flow. There are some cases I need to have a nn.BatchNorm() (BN) standalone, i.e., no Conv layer in the front of BN. 

For the standalone BN, when using PyTorch 1.4 with Vitis AI 1.3, it will not be implemented as depthwise separable conv, but automatically to CPU device (See image below). In addition, there is no errors, or warning during quatization and compile.
[image]

For the standalone BN, when using PyTorch 1.7.1 with Vitis AI 1.4, it will raise an error during compile (See image below).
[image]

BTW, for Conv->BN, I can see it may be fused as one Conv layer and seccessfully implemented on DPU.

hoadv-qh, 2021-08-01:
Hi @FrankXinqi,

What is previous layer of BatchNorm2d in your model?

>it will not be implemented as depthwise separable conv

Did you read Vitis AI User Guide for v1.3 and v1.4? In which page, it says that or just only your expectation?
I think, at first you can try to find something about BatchNorm compilation from the user guide.
Because it may mention some restrictions about BatchNorm to be implemented by CPU/CPU.

Second, if you expect BatchNorm is converted to equivalent *depthwise separable conv*,
then you can try redefine your model with *depthwise separable conv* explicity rather than depends on compiler.

FrankXinqi, 2021-08-01:
Hi @hoadv-qh. Thanks for your answer. However, you misunderstand the most important qeustions. and gives an answer to another direction. 

> What is previous layer of BatchNorm2d in your model?

To your first qeustion, including BN, my layers are 4 relu + 1 concat + 1 BN.

> Did you read Vitis AI User Guide for v1.3 and v1.4? In which page, it says that or just only your expectation?
> I think, at first you can try to find something about BatchNorm compilation from the [...]

To this question. Yes, I have read both 1.3 1.4 ug1414. For 1.3 guide, I use this https://www.xilinx.com/support/documentation/sw_manuals/vitis_ai/1_3/ug1414-vitis-ai.pdf. On page 102, you can find yourself. and I can screenshot the following to you. Please read the right most column. I may misunderstand the depthwise-conv2d there.
[image]

> 
> Second, if you expect BatchNorm is converted to equivalent _depthwise separable conv_,
> then you can try redefine your model with _depthwise separable conv_ explicity rather than depends on compiler.

To this questions. This is a very misunderstand. I only want BN to successfully implemented on DPU. I do not care about if it uses depthwise conv2d or anything else.

To return to the topic, the question is BN is not successfully implemented on DPU. I do NOT care about how it is implemented at this moment.

hoadv-qh, 2021-08-02:
Hi @FrankXinqi,

>To this question. Yes, I have read both 1.3 1.4 ug1414. For 1.3 guide, I use this https://www.xilinx.com/support/documentation/sw_manuals/vitis_ai/1_3/ug1414-vitis-ai.pdf. On page 102,

From the description, it is difficult to understand condition (or when) of  "can be transformed to a depthwise-conv2d equivalently". This is, we need Vitis-AI developer help.

About my 1st and 3rd, It can be for the case that your target is running a task of detecting something on DPU.
If so, you can consider idea of equivalent model/layer which is compatible to vitis-ai as a solution.
But if you have constraint that model must be same as your description, I do not have any idea now.

wanghong4compiler, 2022-02-16:
Hi @FrankXinqi ,
As you can see, for BN, compiler would fuse it to Conv2d or other operators, so for now compiler doesn't support standalone BN.
According to the scene you described, "4 relu + 1 concat + 1 BN", I suggest to move BN to the front of concat, this would not increase the workload, but help improve the performance.
Thank you.

closed 2022-03-03 by qianglin-xlnx