Xilinx/Vitis-AI #495: Can't find xir package in your environment.

wanghao16020510036, 2021-08-02:
I want to install vai_q_pytorch from source code following [_https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Quantizer/vai_q_pytorch#install-from-source-code_](https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Quantizer/vai_q_pytorch#install-from-source-code).    
When I verify the installation, I run this command in my conda env: 
```shell
python -c "import pytorch_nndct"
``` 
And then I got this warning message:
[VAIQ_WARN]: Can't find xir package in your environment.  
I see that XIR library needs to be installed to create deployed model, but how should I do to install xir in my conda env?

niuxjxlnx, 2021-08-05:
@wanghao16020510036 :

You can try

hanxue, 2021-08-05:
Hi @wanghao16020510036 

The XIR conda package is part of the Vitis AI 1.4 release's conda-channel package. You can download it from https://www.xilinx.com/bin/public/openDownload?filename=conda-channel_1.4.914-01.tar.gz

Let's say you want to create a new conda environment with pytorch_nndct, you can use these commands

```
cd /tmp
wget -O conda-channel.tar.gz https://www.xilinx.com/bin/public/openDownload?filename=conda-channel_1.4.914-01.tar.gz
tar zxf conda-channel.tar.gz 
conda create --name nndct-pytorch -c file:///tmp/conda-channel -c pytorch -c conda-forge  pytorch_nndct
```

wanghao16020510036, 2021-08-05:
> Hi @wanghao16020510036
>
> The XIR conda package is part of the Vitis AI 1.4 release's conda-channel package. You can download it from [...]
I tr

closed 2021-08-05 by wanghao16020510036

wanghao16020510036, 2021-08-05:
> Hi @wanghao16020510036
>
> The XIR conda package is part of the Vitis AI 1.4 release's conda-channel package. You can download it from [...]

Thanks for your reply! But **pytorch1.4** and pytorch_nndct1.4 are still installed in conda env "nndct-pytorch". For some special reasons, I must install **pytorch1.6**. How to use pytorch1.6 and pytorch_nndct corresponding to it?

niuxjxlnx, 2021-08-05:
Hi, @wanghao16020510036:

Please try the following steps:

SpursLipu, 2021-09-17:
> Hi, @wanghao16020510036:
> 
> Please try the following steps:
> 
> 1. create a new clean conda env and activate it
> 2. install pytorch 1.6 and pytorch_nndct following the instructions in https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Quantizer/vai_q_pytorch#install-from-source-code.
> 3. get XIR package at https://www.xilinx.com/bin/public/openDownload?filename=conda-channel_1.4.914-01.tar.gz:
> 
> ```
>    cd /tmp
>    wget -O conda-channel.tar.gz https://www.xilinx.com/bin/public/openDownload?filename=conda-channel_1.4.914-01.tar.gz
>    tar zxf conda-channel.tar.gz 
> ```
> 
> 1. install XIR into your conda environment with:
>    conda install xir -c file:///tmp/conda-channel

Hi I ues this way install xir in my own, but when export xmodel, this bug appears.

[VAIQ_NOTE]: =>Converting to xmodel
free(): invalid pointer

chuong98, 2022-06-11:
@SpursLipu I have the same error `free(): invalid pointer`. Did you figure out the solution.