Xilinx/Vitis-AI #499: DPUCADF8H fails for U200 compilation (U280 DPUCAHX8H works fine)

BraunPhilipp, 2021-08-04:
I am trying to compile a model for my U200 card. The model compiles fine for U280 cards using the following command. However, once I compile for U200 with ``DPUCADF8H``` I get a compilation error. I am not entirely sure how to debug this since the error message is quite obscure.

**U280 command:**

````
vai_c_tensorflow --frozen_pb="./quantize/quantize_eval_model.pb" --arch /opt/vitis_ai/compiler/arch/DPUCAHX8H/U280/arch.json --output_dir="compile" --options="{'mode':'normal'}" --net_name "point_pillars"
````

**Output for the U280 compilation:**

````

[INFO] parse raw model     :  0%|          | 0/50 [00:00<?, ?it/s]                           
[INFO] parse raw model     :100%|██████████| 50/50 [00:00<00:00, 30122.84it/s]               

[INFO] infer shape (NHWC)  :  0%|          | 0/50 [00:00<?, ?it/s]                           
[INFO] infer shape (NHWC)  :100%|██████████| 50/50 [00:00<00:00, 77557.40it/s]               

[INFO] infer shape (NHWC)  :  0%|          | 0/50 [00:00<?, ?it/s]                           
[INFO] infer shape (NHWC)  :100%|██████████| 50/50 [00:00<00:00, 19647.29it/s]               

[INFO] generate xmodel     :  0%|          | 0/50 [00:00<?, ?it/s]                           
[INFO] generate xmodel     :100%|██████████| 50/50 [00:00<00:00, 5912.80it/s]                
[INFO] Namespace(inputs_shape=None, layout='NHWC', model_files=['./quantize/quantize_eval_model.pb'], model_type='tensorflow', out_filename='compile/point_pillars_org.xmodel', proto=None)
[INFO] tensorflow model: quantize/quantize_eval_model.pb
[INFO] generate xmodel: /workspace/pointpillars/PointPillars/compile/point_pillars_org.xmodel
[UNILOG][INFO] The compiler log will be dumped at "/tmp/vitis-ai-user/log/xcompiler-20210804-204732-1522"
[UNILOG][INFO] Target architecture: DPUCAHX8H_ISA2
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCAHX8H_ISA2
[UNILOG][INFO] Graph name: quantize_eval_model, with op num: 112
[UNILOG][INFO] Begin to compile...
[UNILOG][INFO] Total device subgraph number 3, DPU subgraph number 1
[UNILOG][INFO] Compile done.
[UNILOG][INFO] The meta json is saved to "/workspace/pointpillars/PointPillars/compile/meta.json"
[UNILOG][INFO] The compiled xmodel is saved to "/workspace/pointpillars/PointPillars/compile/point_pillars.xmodel"
[UNILOG][INFO] The compiled xmodel's md5sum is 14b5bfcf55373890fd88a573f8183b83, and been saved to "/workspace/pointpillars/PointPillars/compile/md5sum.txt"
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
````

**U200 command:**

````
vai_c_tensorflow --frozen_pb="./quantize/quantize_eval_model.pb" --arch /opt/vitis_ai/compiler/arch/DPUCADF8H/U200/arch.json --output_dir="compile" --options="{'mode':'normal'}" --net_name "point_pillars"
````

**Output for U200 compilation:**

````
[INFO] parse raw model     :  0%|          | 0/50 [00:00<?, ?it/s]                           
[INFO] parse raw model     :100%|██████████| 50/50 [00:00<00:00, 30305.66it/s]               

[INFO] infer shape (NHWC)  :  0%|          | 0/50 [00:00<?, ?it/s]                           
[INFO] infer shape (NHWC)  :100%|██████████| 50/50 [00:00<00:00, 76260.07it/s]               

[INFO] infer shape (NHWC)  :  0%|          | 0/50 [00:00<?, ?it/s]                           
[INFO] infer shape (NHWC)  :100%|██████████| 50/50 [00:00<00:00, 19734.19it/s]               

[INFO] generate xmodel     :  0%|          | 0/50 [00:00<?, ?it/s]                           
[INFO] generate xmodel     :100%|██████████| 50/50 [00:00<00:00, 5916.64it/s]                
[INFO] Namespace(inputs_shape=None, layout='NHWC', model_files=['./quantize/quantize_eval_model.pb'], model_type='tensorflow', out_filename='compile/point_pillars_org.xmodel', proto=None)
[INFO] tensorflow model: quantize/quantize_eval_model.pb
[INFO] generate xmodel: /workspace/pointpillars/PointPillars/compile/point_pillars_org.xmodel
[UNILOG][INFO] The compiler log will be dumped at "/tmp/vitis-ai-user/log/xcompiler-20210804-204254-1470"
[UNILOG][INFO] [32mCompiling subgraph : subgraph_cnn/block1/conv2d0/Conv2D[0m
 ############################################# 
 ######  Hyper Graph Construction 
[... 234 lines cut ...]
11 19 cnn/block2/conv2d5/Conv2D ON 2 CUR 4 BY 1
12 19 <class 'SC.HwAbstraction.code_convolution.Conv'> False
12 19 cnn/block3/conv2d0/Conv2D ON 2 CUR 4 BY 1
13 19 <class 'SC.HwAbstraction.code_convolution.Conv'> False
13 19 cnn/block3/conv2d1/Conv2D ON 4 CUR 4 BY 5
14 19 <class 'SC.HwAbstraction.code_convolution.Conv'> False
14 19 cnn/block3/conv2d2/Conv2D ON 4 CUR 4 BY 5
15 19 <class 'SC.HwAbstraction.code_convolution.Conv'> False
15 19 cnn/block3/conv2d3/Conv2D ON 4 CUR 4 BY 5
16 19 <class 'SC.HwAbstraction.code_convolution.Conv'> False
16 19 cnn/block3/conv2d4/Conv2D ON 4 CUR 4 BY 4
17 19 <class 'SC.HwAbstraction.code_end.Bracket_End'> False
17 19 cnn/block3/conv2d5/Conv2D ON 4 CUR 4 BY 2
18 19 bracket ON 2 CUR 2 BY 0
 ############################################# 
 ######  Code Generation at Node Level and then Recursively
 ############################################# 
CODE cnn/block1/conv2d0/Conv2D conv2d
BATCH IN  Shape [1, 504, 504, 64] Heights [15, 14, 14, 14, 14, 14, 14, 14, 14, 14, 14, 14, 14, 14, 14, 14, 14, 14, 14, 14, 14, 14, 14, 14, 14, 14, 14, 14, 14, 14, 14, 14, 14, 14, 14, 13] 
BATCH OUT Shape [1, 252, 252, 64] Heights [7, 7, 7, 7, 7, 7, 7, 7, 7, 7, 7, 7, 7, 7, 7, 7, 7, 7, 7, 7, 7, 7, 7, 7, 7, 7, 7, 7, 7, 7, 7, 7, 7, 7, 7, 7] 
'NoneType' object has no attribute 'tensor'
> /opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/SC/HwAbstraction/code_generation.py(458)my_simplified_code()
-> CCode.extend(c.code_generation(dbon,dbby) )
(Pdb) 
````

paolodalberto, 2021-08-12:
interesting you hit a bug in the compiler for DPUV3INT8 
is it possible to share the model ? The model looks like a sequence of convolutions like "VGG".
if it is not possible ... you reached a debug trace, please let it run further: please type "c" and show us the trace. 
The problem is related to the "first layer" ...  thank you

BraunPhilipp, 2021-08-12:
Thank you for reaching out! The output above was generated with a [patched compiler from Vitis AI 1.3](https://github.com/Xilinx/Vitis-AI/blob/06807224282eee49042cd48538e49d3e6db9142b/docs/quick-start/install/Vitis%20AI%201.3.2%20April%202021%20Patch.md).

In the most recent version from Vitis AI; I get a slightly different output as seen below. The U280 DPU still compiles fine.

The model I am using is part of a custom PointPillars model. The subgraph from the model was derived using a custom partitioner. The protobuf after partitioning is in the following zip folder ([center_frozen.zip](https://github.com/Xilinx/Vitis-AI/files/6978930/center_frozen.zip)).

Definitely reach out via email if further details are helpful. 

```
[INFO] Namespace(batchsize=1, inputs_shape=None, layout='NHWC', model_files=['./output_new_quantize/quantize_eval_model.pb'], model_type='tensorflow', named_inputs_shape=None, out_filename='/tmp/point_pillars_org.xmodel', proto=None)
[INFO] tensorflow model: /workspace/pointpillars/PointPillars/output_new_quantize/quantize_eval_model.pb

[INFO] parse raw model     :  0%|          | 0/50 [00:00<?, ?it/s]                           
[INFO] parse raw model     :100%|██████████| 50/50 [00:00<00:00, 29396.58it/s]               

[INFO] infer shape (NHWC)  :  0%|          | 0/50 [00:00<?, ?it/s]                           
[INFO] infer shape (NHWC)  : 28%|██▊       | 14/50 [00:00<00:00, 137.32it/s]                 
[INFO] infer shape (NHWC)  :100%|██████████| 50/50 [00:00<00:00, 317.32it/s]                 

[INFO] perform level-0 opt :  0%|          | 0/2 [00:00<?, ?it/s]                            
[INFO] perform level-0 opt :100%|██████████| 2/2 [00:00<00:00, 1245.15it/s]                  

[INFO] perform level-1 opt :  0%|          | 0/6 [00:00<?, ?it/s]                            
[INFO] perform level-1 opt :100%|██████████| 6/6 [00:00<00:00, 1236.71it/s]                  

[INFO] generate xmodel     :  0%|          | 0/50 [00:00<?, ?it/s]                           
[INFO] generate xmodel     :100%|██████████| 50/50 [00:00<00:00, 7547.24it/s]                
[INFO] dump xmodel ...
[INFO] dump xmodel: /tmp/point_pillars_org.xmodel
[UNILOG][INFO] Target architecture: DPUCADF8H_ISA0
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCADF8H_ISA0
[UNILOG][INFO] Graph name: quantize_eval_model, with op num: 112
[UNILOG][INFO] Begin to compile...
[0;33m[UNILOG][WARNING] DPU prefers xir::Op{name = cnn/block1/conv2d0/Conv2D, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[m[0;33m[UNILOG][WARNING] xir::Op{name = cnn/block1/conv2d0/Conv2D, type = conv2d-fix} has been assigned to CPU.
[m[0;33m[UNILOG][WARNING] DPU prefers xir::Op{name = cnn/block1/conv2d1/Conv2D, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[m[0;33m[UNILOG][WARNING] xir::Op{name = cnn/block1/conv2d1/Conv2D, type = conv2d-fix} has been assigned to CPU.
[m[0;33m[UNILOG][WARNING] DPU prefers xir::Op{name = cnn/block1/conv2d2/Conv2D, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[m[0;33m[UNILOG][WARNING] xir::Op{name = cnn/block1/conv2d2/Conv2D, type = conv2d-fix} has been assigned to CPU.
[m[0;33m[UNILOG][WARNING] DPU prefers xir::Op{name = cnn/block1/conv2d3/Conv2D, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[m[0;33m[UNILOG][WARNING] xir::Op{name = cnn/block1/conv2d3/Conv2D, type = conv2d-fix} has been assigned to CPU.
[m[0;33m[UNILOG][WARNING] DPU prefers xir::Op{name = cnn/block2/conv2d0/Conv2D, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[m[0;33m[UNILOG][WARNING] xir::Op{name = cnn/block2/conv2d0/Conv2D, type = conv2d-fix} has been assigned to CPU.
[m[0;33m[UNILOG][WARNING] DPU prefers xir::Op{name = cnn/block2/conv2d1/Conv2D, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[m[0;33m[UNILOG][WARNING] xir::Op{name = cnn/block2/conv2d1/Conv2D, type = conv2d-fix} has been assigned to CPU.
[m[0;33m[UNILOG][WARNING] DPU prefers xir::Op{name = cnn/block2/conv2d2/Conv2D, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[m[0;33m[UNILOG][WARNING] xir::Op{name = cnn/block2/conv2d2/Conv2D, type = conv2d-fix} has been assigned to CPU.
[m[0;33m[UNILOG][WARNING] DPU prefers xir::Op{name = cnn/block2/conv2d3/Conv2D, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[m[0;33m[UNILOG][WARNING] xir::Op{name = cnn/block2/conv2d3/Conv2D, type = conv2d-fix} has been assigned to CPU.
[m[0;33m[UNILOG][WARNING] DPU prefers xir::Op{name = cnn/block2/conv2d4/Conv2D, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[m[0;33m[UNILOG][WARNING] xir::Op{name = cnn/block2/conv2d4/Conv2D, type = conv2d-fix} has been assigned to CPU.
[m[0;33m[UNILOG][WARNING] DPU prefers xir::Op{name = cnn/block2/conv2d5/Conv2D, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[m[0;33m[UNILOG][WARNING] xir::Op{name = cnn/block2/conv2d5/Conv2D, type = conv2d-fix} has been assigned to CPU.
[m[0;33m[UNILOG][WARNING] DPU prefers xir::Op{name = cnn/block3/conv2d0/Conv2D, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[... cut ...]
[UNILOG][INFO] The meta json is saved to "/workspace/pointpillars/PointPillars/output_test/meta.json"
[UNILOG][INFO] The compiled xmodel is saved to "/workspace/pointpillars/PointPillars/output_test/point_pillars.xmodel"
[UNILOG][INFO] The compiled xmodel's md5sum is d318572e3570750ffbb0505724e70ce2, and has been saved to "/workspace/pointpillars/PointPillars/output_test/md5sum.txt"
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************

```

paolodalberto, 2021-08-13:
the good news is that 1.4 does better. 
let me see if I can retrieve and compile  your model today

paolodalberto, 2021-08-13:
what is the input shape @BraunPhilipp ?
I cannot parse the pb file ... I will need help internally to transform the pb into an acceptable xmodel (internal XIR)

paolodalberto, 2021-08-13:
I will use the shape [4,504,504,64]... nope I Cannot parse it either. 
it is missing all the information about quantization and the model is a pb but it looks like a caffe :)
Let me see what it will look like

BraunPhilipp, 2021-08-13:
Thank you for your effort. Input shape is [1, 504, 504, 64] and is usually easy to find through graph explorers like Netron. The quantized model is in the following zip folder. [quantize_eval_model.zip](https://github.com/Xilinx/Vitis-AI/files/6983877/quantize_eval_model.zip). The original model is created through keras.

paolodalberto, 2021-08-13:
currently a proficient way to use "dpuv3int8" is with batch=4
let me compile the quantized one

paolodalberto, 2021-08-13:
I can 
```
 #############################################
 ######  Success boost
 #############################################
 #############################################
 ######  Writing code to file:work/out.asm
 #############################################
 BUFFER SIZE and RELATIVE ADDRESS
 PARAMETER  BUFFER SIZE 0x1ca000 and RELATIVE ADDRESS 0x0
 INPUT      BUFFER SIZE 0x3e04000 and RELATIVE ADDRESS 0x1ca000
 OUTPUT BUFFER SIZE 0x1f0800 and RELATIVE ADDRESS 0x3fce000
 SWAP BUFFER SIZE 0x1f02000 and RELATIVE ADDRESS 0x41be800
 INSTRUCTION BUFFER SIZE 0x40 and RELATIVE ADDRESS 0x30000000
 BUFFER SIZE and RELATIVE ADDRESS

    REG_ID_FORMATTED = 0
    REG_RESULT_SPACE = 1
    REG_SWAP_SPACE   = 2
    REG_PAR_SPACE    = 3

    ADD_REG_ID_FORMATTED = 0x10000000
    ADD_REG_RESULT_SPACE = 0x10000000
    ADD_REG_SWAP_SPACE   = 0x10000000
    ADD_REG_PAR_SPACE    = 0x10000000

    druDstBufSize: 16777216,

 #############################################
 ######  #### JSON
 #############################################
{'inputs': {'input_1': {'name': 'input_1', 'name_in_fpga': 'input_1/aquant', 'shape': TensorShapes(batch=4, height=504, width=504, channel=$
 #############################################
 ######  Garbage 0
 #############################################
Type      Number              Operations          Duration
----------------------------------------------------------------------
LOADFM    2276                179343360           11531.851851851206
LOADW     32                  1850368             118.97942386831278
LOADTH    0                   0                   0
SAVE      1827                115831296           7447.99999999976
CONV      276                 206009008128        44393.56701030927
ELEW      0                   0                   0
POOL      0                   0                   0
----------------------------------------------------------------------
Total Time = 45342.47 us (FPS = 352.87)
```

paolodalberto, 2021-08-13:
the flow looks good:
[image]

paolodalberto, 2021-08-13:
let me ask if we can reproduce your problem on our side (using the xcompiler interface) because I cannot reproduce it using my tools ...

paolodalberto, 2021-08-13:
ok, it looks like we can run on FPGA ...

paolodalberto, 2021-08-13:
Such a pity that we can run it in the sand box but not in the wild.
Now that I look at your output it looks like there 
```

�[m[UNILOG][INFO] Total device subgraph number 2, DPU subgraph number 0
[UNILOG][INFO] Compile done.
[UNILOG][INFO] The meta json is saved to "/workspace/pointpillars/PointPillars/output_test/meta.json"
[UNILOG][INFO] The compiled xmodel is saved to "/workspace/pointpillars/PointPillars/output_test/point_pillars.xmodel"
[UNILOG][INFO] The compiled xmodel's md5sum is d318572e3570750ffbb0505724e70ce2, and has been saved to "/workspace/pointpillars/PointPillars/output_test/md5sum.txt"
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
```
there are no DPU graphs .... 
this is off from my experience

paolodalberto, 2021-08-13:
@BraunPhilipp  can you attach the output of the successful compilation ?
Can you use batch 4 ?

BraunPhilipp, 2021-08-14:
Interestingly using a batch size of 4 results in the same error for the DPUCADF8H. The quantized model can be found in 
[quantized_batch_4.zip](https://github.com/Xilinx/Vitis-AI/files/6986493/quantized_batch_4.zip). The output for batch size 4 and DPUCADF8H is the following. Note that the compilation finishes successful and generates an xmodel file in Vitis 1.4 despite the errors or warnings seen below.

````
[INFO] Namespace(batchsize=1, inputs_shape=None, layout='NHWC', model_files=['./output_new_quantize/quantize_eval_model.pb'], model_type='tensorflow', named_inputs_shape=None, out_filename='/tmp/point_pillars_org.xmodel', proto=None)
[INFO] tensorflow model: /workspace/pointpillars/PointPillars/output_new_quantize/quantize_eval_model.pb

[INFO] parse raw model     :  0%|          | 0/50 [00:00<?, ?it/s]                           
[INFO] parse raw model     :100%|██████████| 50/50 [00:00<00:00, 29793.32it/s]               

[INFO] infer shape (NHWC)  :  0%|          | 0/50 [00:00<?, ?it/s]                           
[INFO] infer shape (NHWC)  : 36%|███▌      | 18/50 [00:00<00:00, 176.40it/s]                 
[INFO] infer shape (NHWC)  :100%|██████████| 50/50 [00:00<00:00, 348.64it/s]                 

[INFO] perform level-0 opt :  0%|          | 0/2 [00:00<?, ?it/s]                            
[INFO] perform level-0 opt :100%|██████████| 2/2 [00:00<00:00, 1240.55it/s]                  

[INFO] perform level-1 opt :  0%|          | 0/6 [00:00<?, ?it/s]                            
[INFO] perform level-1 opt :100%|██████████| 6/6 [00:00<00:00, 1239.33it/s]                  

[INFO] generate xmodel     :  0%|          | 0/50 [00:00<?, ?it/s]                           
[INFO] generate xmodel     :100%|██████████| 50/50 [00:00<00:00, 7754.88it/s]                
[INFO] dump xmodel ...
[INFO] dump xmodel: /tmp/point_pillars_org.xmodel
[UNILOG][INFO] Target architecture: DPUCADF8H_ISA0
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCADF8H_ISA0
[UNILOG][INFO] Graph name: quantize_eval_model, with op num: 112
[UNILOG][INFO] Begin to compile...
[0;33m[UNILOG][WARNING] DPU prefers xir::Op{name = cnn/block1/conv2d0/Conv2D, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[m[0;33m[UNILOG][WARNING] xir::Op{name = cnn/block1/conv2d0/Conv3D, type = conv2d-fix} has been assigned to CPU.
[m[0;33m[UNILOG][WARNING] DPU prefers xir::Op{name = cnn/block1/conv2d1/Conv2D, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[m[0;33m[UNILOG][WARNING] xir::Op{name = cnn/block1/conv2d1/Conv2D, type = conv2d-fix} has been assigned to CPU.
[m[0;33m[UNILOG][WARNING] DPU prefers xir::Op{name = cnn/block1/conv2d2/Conv2D, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[m[0;33m[UNILOG][WARNING] xir::Op{name = cnn/block1/conv2d2/Conv2D, type = conv2d-fix} has been assigned to CPU.
[m[0;33m[UNILOG][WARNING] DPU prefers xir::Op{name = cnn/block1/conv2d3/Conv2D, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[m[0;33m[UNILOG][WARNING] xir::Op{name = cnn/block1/conv2d3/Conv2D, type = conv2d-fix} has been assigned to CPU.
[m[0;33m[UNILOG][WARNING] DPU prefers xir::Op{name = cnn/block2/conv2d0/Conv2D, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[m[0;33m[UNILOG][WARNING] xir::Op{name = cnn/block2/conv2d0/Conv2D, type = conv2d-fix} has been assigned to CPU.
[m[0;33m[UNILOG][WARNING] DPU prefers xir::Op{name = cnn/block2/conv2d1/Conv2D, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[m[0;33m[UNILOG][WARNING] xir::Op{name = cnn/block2/conv2d1/Conv2D, type = conv2d-fix} has been assigned to CPU.
[m[0;33m[UNILOG][WARNING] DPU prefers xir::Op{name = cnn/block2/conv2d2/Conv2D, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[m[0;33m[UNILOG][WARNING] xir::Op{name = cnn/block2/conv2d2/Conv2D, type = conv2d-fix} has been assigned to CPU.
[m[0;33m[UNILOG][WARNING] DPU prefers xir::Op{name = cnn/block2/conv2d3/Conv2D, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[m[0;33m[UNILOG][WARNING] xir::Op{name = cnn/block2/conv2d3/Conv2D, type = conv2d-fix} has been assigned to CPU.
[m[0;33m[UNILOG][WARNING] DPU prefers xir::Op{name = cnn/block2/conv2d4/Conv2D, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[m[0;33m[UNILOG][WARNING] xir::Op{name = cnn/block2/conv2d4/Conv2D, type = conv2d-fix} has been assigned to CPU.
[m[0;33m[UNILOG][WARNING] DPU prefers xir::Op{name = cnn/block2/conv2d5/Conv2D, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[m[0;33m[UNILOG][WARNING] xir::Op{name = cnn/block2/conv2d5/Conv2D, type = conv2d-fix} has been assigned to CPU.
[m[0;33m[UNILOG][WARNING] DPU prefers xir::Op{name = cnn/block3/conv2d0/Conv2D, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[m[0;33m[UNILOG][WARNING] xir::Op{name = cnn/block3/conv2d0/Conv2D, type = conv2d-fix} has been assigned to CPU.
[m[0;33m[UNILOG][WARNING] DPU prefers xir::Op{name = cnn/block3/conv2d1/Conv2D, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.
[... cut ...]
[UNILOG][INFO] The meta json is saved to "/workspace/pointpillars/PointPillars/output_new_quantize/meta.json"
[UNILOG][INFO] The compiled xmodel is saved to "/workspace/pointpillars/PointPillars/output_new_quantize/point_pillars.xmodel"
[UNILOG][INFO] The compiled xmodel's md5sum is 5ecbd03479bbb758e1c87b4948a954f5, and has been saved to "/workspace/pointpillars/PointPillars/output_new_quantize/md5sum.txt"
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
````

BraunPhilipp, 2021-08-14:
How did you generate the flow below?

I am still getting familiar with the toolchain from Xilinx. Any tools that are publicly available?

> the flow looks good:
> [image]

paolodalberto, 2021-08-16:
Currently, we can run your model in FPGA internally: I do not know when the current code will be released and I do not know why you cannot run on your side. The output for both models does not tell me anything about the failure. Highly suspicious  the message about zero DPU subgraphs.  There must be one!

This could be because the partitioner did not recognize the graph properly. This unfortunately I cannot help and my colleague could actually run graph on the FPGA.  I understand, you cannot run and the frustrating part is that we can using. A new release, very likely will fix your problems.

The graphical tool is not available, not clear why, because there will be even more stuff to explain and how to use it. This is a tool the compiler uses to debug synchronizations issues (those you do not have). 

How can we move on from here ?
I do not know

BraunPhilipp, 2021-08-16:
Thank you for your effort! I am now using the U280 so that is not an issue. 

It would be amazing if you could open source the graphical tool. It would be really useful to debugging purposes.

closed 2021-08-16 by BraunPhilipp

paolodalberto, 2021-08-16:
thank you for the submission

paolodalberto, 2021-08-16:
did you notice if the batch = 4 is truly batch 4 ?
`�[0;33m[UNILOG][WARNING] DPU prefers xir::Op{name = cnn/block1/conv2d0/Conv2D, type = conv2d-fix}'s input batch to be 4, but it's 1 now. So it will be assigned to CPU.`

paolodalberto, 2021-08-16:
it looks like you have still a batch = 1 ... before we move on ... can we check it ? @BraunPhilipp

woinck, 2021-08-16:
Could you share the code you used for compiling U200 models? To enable batchsize 4 compilation, please try adding a `input_shape` flag to the vai_c_tensorflow options like this:
`vai_c_tensorflow --frozen_pb="./quantize/quantize_eval_model.pb" --arch /opt/vitis_ai/compiler/arch/DPUCADF8H/U200/arch.json --output_dir="compile" --options="{'mode':'normal', 'input_shape': '4,224,224,3'}" --net_name "point_pillars"`
(replace the 224x224 input shape with your model's input shape)

paolodalberto, 2021-08-16:
Thank you @woinck

paolodalberto, 2021-08-17:
@BraunPhilipp  any help ?

BraunPhilipp, 2021-08-18:
Thank you @woinck for the support! Adding `input_shape` as an option fixed the issue. Apparently, the default is set to a batch size of 1 despite no support.

@paolodalberto Is there any detailed documentation for the quantizer and/or compiler? I faced other issues with the compiler for transposed convolutions and would like to patch it.

paolodalberto, 2021-08-18:
@BraunPhilipp 
The good news is that the future deployment the default batch does not matter (we should be able to manage batch =1). 
There should be documentation for the model zoo and batch= 4 but I am not sure (sorry). The compiler for 1.4 takes a hardline approach and refuses  to compile for batch !=4.  

Transpose convolution = deconvolution  is supported.  if you have any problem, please create an issue here and add in the comment me @paolodalberto  so I get a notification and can follow up with you

BraunPhilipp, 2021-08-18:
Perfect! Thank you again for the support!