Xilinx/Vitis-AI #500: About pytorch version in Vitis-AI-1.4

wanghao16020510036, 2021-08-05:
It is mentioned that *support pytorch 1.5-1.7.1* in [readme](https://github.com/Xilinx/Vitis-AI/tree/master). But pytorch1.4 and pytorch_nndct1.4 are still installed in conda env "vitis-ai-pytorch" in docker container.  
For some special reasons, I must install pytorch1.6. It is obvious that arbitrarily updating pytorch to version 1.6 is not feasible, because it depends on cuda, cudnn and pytorch_nndct, etc. How to use pytorch1.6 and pytorch_nndct corresponding to it? Looking forward to anyone's help.

Enoch2090, 2021-08-08:
See [Vitis-AI-Quantizer/vai_q_pytorch#supported-and-limitaion](https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Quantizer/vai_q_pytorch#supported-and-limitaion). You can use `/opt/vitis_ai/scripts/replace_pytorch.sh` to change the version of PyTorch.

niuxjxlnx, 2022-02-21:
Pytorch 1.7 version has been supported in Vitis-AI 1.4, and pytorch 1.8, 1.9 version have been supported in Vitis-AI 2.0.
Further more, solution to change pytorch versions among the supported version range is released, please refer to the related part about script replace_pytorch.sh in user guide.

closed 2022-02-21 by niuxjxlnx