Xilinx/Vitis-AI #501: Facing issue with neptune

durgabhavaniv, 2021-08-05:
I am trying to run Neptune without any errors, but am not able to get output in the browser. [https://github.com/Xilinx/Vitis-AI/tree/v1.3.2/demo/neptune](https://github.com/Xilinx/Vitis-AI/tree/v1.3.2/demo/neptune)
Output log:
[image]
Bowser log:
[image]

varunsh-xilinx, 2021-08-17:
I don't see what errors you got but I got some missing package errors when I tried it using some 1.3.* containers. It looks like there are some conflicts in conda in the newer versions of the docker container that interfere with Neptune. However, I was able to start the server with an earlier version of Vitis-AI.

```bash
# on host, clone the Vitis-AI repo
$ cd Vitis-AI
$ git checkout v1.1
$ ./docker_run.sh xilinx/vitis-ai-cpu:1.1.56

# in container
$ conda activate vitis-ai-neptune
$ cd alveo/neptune
$ ./run.sh
```

The server GUI should be visible in the browser on port 8998.

renjiangyong, 2022-09-06:
@durgabhavaniv @varunsh-xilinx  , you can use Vitis-AI 2.5 for experiment.  The issue is open for a long time, I will close it, you can reopen it when you need.

closed 2022-09-06 by renjiangyong