Xilinx/Vitis-AI #502: Facing issues when compiling Flatten and Fully Connected (Dense) layers.

liunett, 2021-08-06:
I am trying to run a model with just Flatten and Fully Connected layers on an Ultra96-V2 device but I'm facing and unexpected reshaping of the input layer. With the Digits Mnist dataset, which are (28, 28, 1) images, with only Flatten and Dense layers the compiled input turned into (4, 14, 14).

I've seen that the Vitis AI Compiler automatically combines the FC and Flatten layers to a Conv2D layer, is there any public information about how that reshaping is done?

I've seen that (2, 2, 1) numpy arrays turn into (1, 1, 4) arrays, (4, 4, 1) ones turn into (4, 2, 2) and (6, 6, 1) turn into (6, 2, 3).
Some details about how this reshaping is done to generalize my data preprocessing would be appreciated, or knowing how this problem is usually faced would be appreciated aswell.

qianglin-xlnx, 2021-08-10:
Hi @liunett 
Which version of Vitis AI tool are you using? And could you share the quantized model ? Thank you.

liunett, 2021-08-10:
> Hi @liunett
> Which version of Vitis AI tool are you using? And could you share the quantized model ? Thank you.

Docker Image Version:  1.4.916
Quantized Model: https://anonfiles.com/h0BfteBcu6/quantized_model_h5

On the quantized model summary you won't be able to see the reshaping yet, the  shape of the dpu input tensors is where you can check it.

wanghong4compiler, 2022-02-16:
Hi @liunett 
Sorry for replying so late. the quantized model file has been expired, could you upload this again?
so that I can reproduce this issue and get the details.
thank you very much.

qianglin-xlnx, 2022-03-03:
hi @liunett 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-03-03 by qianglin-xlnx