Xilinx/Vitis-AI #503: Error quantizing PyTorch YOLOv4

Enoch2090, 2021-08-08:
I am trying to compile [Tianxiaomo/pytorch-YOLOv4](https://github.com/Tianxiaomo/pytorch-YOLOv4) using Vitis-ai Docker Image 1.4.916. Here's the log:
```bash
(vitis-ai-pytorch) Vitis-AI /workspace/yolov4 > ./build.sh 
No CUDA runtime is found, using CUDA_HOME='/usr/local/cuda'

[VAIQ_NOTE]: Loading NNDCT kernels...
-------- Start Yolov4 test 

[VAIQ_NOTE]: Quantization calibration process start up...

[VAIQ_NOTE]: =>Quant Module is in 'cpu'.

[VAIQ_NOTE]: =>Parsing Yolov4...
/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/numpy/core/function_base.py:22: TracerWarning: Converting a tensor to a Python index might cause the trace to be incorrect. We can't record the data flow of Python values, so this value will be treated as a constant in the future. This means that the trace might not generalize to other inputs!
  i = operator.index(i)
/workspace/yolov4/tool/yolo_layer.py:227: TracerWarning: torch.tensor results are registered as constants in the trace. You can safely ignore this warning if you use this function to create tensors out of constant variables that would be the same every time you call this function. In any other case, this might cause the trace to be incorrect.
  bx = bxy[:, ii : ii + 1] + torch.tensor(grid_x, device=device, dtype=torch.float32) # grid_x.to(device=device, dtype=torch.float32)
/workspace/yolov4/tool/yolo_layer.py:229: TracerWarning: torch.tensor results are registered as constants in the trace. You can safely ignore this warning if you use this function to create tensors out of constant variables that would be the same every time you call this function. In any other case, this might cause the trace to be incorrect.
  by = bxy[:, ii + 1 : ii + 2] + torch.tensor(grid_y, device=device, dtype=torch.float32) # grid_y.to(device=device, dtype=torch.float32)

[VAIQ_WARN]: The quantizer ignores new op `aten::softplus` by default.

[VAIQ_NOTE]: You can make these new ops quantizable by add them to custom_quant_ops, e.g. quantizer= torch_quantizer(..., custom_quant_ops=['aten::softplus',...])

[VAIQ_NOTE]: =>Doing weights equalization...

[VAIQ_NOTE]: =>Quantizable module is generated.(quantize_result/Yolov4.py)

[VAIQ_NOTE]: Convert `aten::softplus` to quantizable op.
Traceback (most recent call last):
  File "yolov4_quant.py", line 295, in <module>
    file_path=file_path)
  File "yolov4_quant.py", line 223, in quantization
    quant_mode, model, (example_input), device=device, custom_quant_ops=['aten::softplus'])
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/apis.py", line 77, in __init__
    custom_quant_ops = custom_quant_ops)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py", line 130, in __init__
    quantizer.setup(graph, False, lstm_app, custom_quant_ops=custom_quant_ops)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/quantization/base_quantizer.py", line 82, in setup
    custom_quant_ops=custom_quant_ops)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/quantization/quant_info.py", line 42, in __init__
    self._quant_info = quant_strategy.create_quant_config(self)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/quantization/quant_strategy.py", line 138, in create_quant_config
    return self._get_default_quant_config(quant_info_mgr)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/quantization/quant_strategy.py", line 113, in _get_default_quant_config
    end = quant_info_mgr.quant_output(p_n.name).name
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/nndct_graph/nndctgraph_holder.py", line 250, in quant_output
    end_node = self.__Nndctgraph.node(self._QuantGroups[node.name][idx])
IndexError: list index out of range
```
I think I can use torch.jit.trace to find the problem of the three warnings. However I'm not sure what happens for this IndexError. If several `print`s inserted to `/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/nndct_graph/nndctgraph_holder.py`, it can be further investigated that it is node `Yolov4::Yolov4/Yolov4Head[head]/YoloLayer[yolo1]/3445` that triggered this error, where the code wants to access the -3 element of a length-2-list. This node has `op.type` as `strided_slice` which is in the defined list
```python
self.MULTIPLE_OUTPUTS_OPS = [ # OP types where cannot do quantization 
        NNDCT_OP.CHUNK, # has multiple outputs and cannot be deployed
        NNDCT_OP.STRIDED_SLICE, # copy part of memory from a tensor and create a new tensor
        NNDCT_OP.PERMUTE, # can not be deployed
        NNDCT_OP.FLATTEN, # no calculation and only tensor in-place operation
        NNDCT_OP.RESHAPE, # no calculation and only tensor in-place operation
        NNDCT_OP.PIXEL_SHUFFLE, # no calculation and only tensor in-place operation
        NNDCT_OP.CONTIGUOUS, # no calculation and only tensor in-place operation
        NNDCT_OP.SQUEEZE, # no calculation and only tensor in-place operation
    ]
```
and hence the while loop didn't stop. The chunk of code around the error is
```python
  def quant_output(self, node_or_name):
    node = self._find_node(node_or_name)
    idx = -1
    end_node = self.__Nndctgraph.node(self._QuantGroups[node.name][idx])
    if self.is_concat_input(end_node):
      for c in self.Nndctgraph.children(end_node):
        if c.op.type == NNDCT_OP.CONCAT:
          return c
   
    while end_node.op.type in self.MULTIPLE_OUTPUTS_OPS:
      idx = idx - 1
      opt = end_node.op.type
      try:      
        end_node = self.__Nndctgraph.node(self._QuantGroups[node.name][idx])
      except Exception as e:
        print(f"name: {node.name}, idx: {idx}, type: {opt}")
        print(e)
        raise(e)
    return end_node
```
And for most nodes passing through the `while` loop, I found the trace is usually one of the three:
- `[]`
- `['reshape']`
- `['reshape', 'permute', 'reshape']`
For your convenience to reproduce this problem, attached in the gist is the quantization snippet that I used. The command is
```bash
python3 yolov4_quant.py \
--quant_mode calib \
--model_dir "./models" \
--batch_size 1
```
Please point out if I made any mistake. Thanks!

jimheaton, 2021-08-09:
Not related to the error you are seeing, but keep in mind that the DPU does support mish activation. You will need to change to leaky relu and retrain your model. 

See https://github.com/Xilinx/Vitis-Tutorials/tree/master/Machine_Learning/Design_Tutorials/07-yolov4-tutorial for information on how to implement using Darknet to TF or Darknet to Caffe Flow.

Enoch2090, 2021-08-29:
A comment after tackling the problem - The key is to check through every module used in the network to see if it is supported as the ug1414 doc states. For this specific implementation, do the following:
1. Change all Mish activation to LeakyRELU;
2. Replace the Upsample with PyTorch’s builtin torch.nn.upsample, and use fixed input/output sizes instead of dynamically acquiring the size with size().
3. Remove the final YoloLayer, do that outside the DPU.
This should do it.

Michael-Schlosser-SE, 2021-11-11:
@Enoch2090 Do you have an example how the model should look like after these modifications? I am also trying to quantize this model but am relatively new to PyTorch.

niuxjxlnx, 2022-02-21:
Hi, @Enoch2090:

The error "IndexError: list index out of range" you found is caused by a bug in quantizer code. This bug has been fixed Vitis-AI 1.4.1 and 2.0 version. You can use Vitis-AI 1.4.1/2.0 or get the code with fix at https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Quantizer/vai_q_pytorch.

closed 2022-03-17 by hanxue