Xilinx/Vitis-AI #505: commit id for“fa49b842f283242091476cf8e1ae4d242a2a838e”

simplelins, 2021-08-11:
where can i find the source code of commit id 'fa49b842f283242091476cf8e1ae4d242a2a838e'.  I had compiled the vart debug libraries, but the libvart-runner.so.1 caused `Segmentation fault (core dumped)` ,    
when I remove this debug, the demo can run successfully, 
I had got some information about lib version. bug now I can not find the source code of commit id 'fa49b842f283242091476cf8e1ae4d242a2a838e'
 ```
 /home/lww/usr/.local/Ubuntu.18.04.x86_64.Debug/bin/vart_version
Symbol xilinx_version not found in libunilog.so.1: /home/lww/usr/.local/Ubuntu.18.04.x86_64.Debug/lib/libunilog.so.1: undefined symbol: xilinx_version
Symbol xilinx_version not found in libtarget-factory.so.1: /home/lww/usr/.local/Ubuntu.18.04.x86_64.Debug/lib/libtarget-factory.so.1: undefined symbol: xilinx_version
libxir.so.1: Xilinx xir Version: xir-84798c76e6ebb93300bf384cb56397f214676330 2021-08-10-15:53:40
libvart-buffer-object.so.1: Xilinx vart-buffer-object Version: 1.4.0-84798c76e6ebb93300bf384cb56397f214676330  2021-08-10-15:52:49
libvart-dpu-controller.so.1: Xilinx vart-dpu-controller Version: 1.4.0-84798c76e6ebb93300bf384cb56397f214676330  2021-08-10-15:52:49
libvart-dpu-runner.so.1: Xilinx vart-dpu-runner Version: 1.4.0-84798c76e6ebb93300bf384cb56397f214676330  2021-08-10-15:52:49
libvart-mem-manager.so.1: Xilinx vart-mem-manager Version: 1.4.0-84798c76e6ebb93300bf384cb56397f214676330  2021-08-10-15:52:49
libvart-runner.so.1: Xilinx vart-runner Version: 1.4.0-fa49b842f283242091476cf8e1ae4d242a2a838e  2021-07-09-05:11:49
libvart-util.so.1: Xilinx vart-util Version: 1.4.0-84798c76e6ebb93300bf384cb56397f214676330  2021-08-10-15:52:49
libvart-xrt-device-handle.so.1: Xilinx vart-xrt-device-handle Version: 1.4.0-84798c76e6ebb93300bf384cb56397f214676330  2021-08-10-15:52
 ```
the lib 'libvart-dpu-runner.so.1' is coming from `/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/libvart-runner.so.1 (0x00007f113e9ab000)`
please help

qianglin-xlnx, 2021-08-16:
Hi @simplelins 
You can find the source code in https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Runtime/VART/vart 
It's same with commit id 'fa49b842f283242091476cf8e1ae4d242a2a838e'
Commit id 'fa49b842f283242091476cf8e1ae4d242a2a838e' is internal vart repo's commit id and the docker's VART is built from internal vart repo.  That's why you see the commit id for vart in docker, but not on the Github.

simplelins, 2021-08-17:
@qianglin-xlnx 
  Hi， are you sure the  the source code in https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Runtime/VART/vart is same with the commit id  'fa49b842f283242091476cf8e1ae4d242a2a838e'?
  I had rebuild the  'libvart-runner.so.1' for debug, but that caused coredump, 
  please give a check!

qianglin-xlnx, 2021-08-17:
Hi @simplelins 
I have double check the source code in https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Runtime/VART/vart is same with the commit id 'fa49b842f283242091476cf8e1ae4d242a2a838e' except "dpuv1-runner", "sim-runner" and "cpu-runner" are not released.
So which target board are you running?  And have you try building the whole vart in debug and run the vart sample with the debug vart?

closed 2021-08-18 by simplelins