Xilinx/Vitis-AI #507: Error when compiling quantized LSTM with vai_c_xir

chtapodi, 2021-08-11:
I am attempting to compile a quantized LSTM model for the ZCU102, and am getting a fatal error.

`(vitis-ai-pytorch) Vitis-AI /workspace/tools/RNN/rnn_compiler > vai_c_xir -x quantize_result/xmodel/Lstm_StandardLstmCell_layer_0_forward_int.xmodel -a /opt/vitis_ai/compiler/arch/DPUCZDX8G/ZCU102/arch.json -n LSTM
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA0_B4096_MAX_BG2       
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA0_B4096_MAX_BG2       
[UNILOG][INFO] Graph name: Lstm_StandardLstmCell_layer_0_forward, with 
op num: 80[UNILOG][INFO] Begin to compile...[UNILOG][WARNING] xir::Op{name = Lstm_StandardLstmCell_layer_0_forward__ft_input_2_new, type = depthwise-fix} has been assigned to CPU: [DPUCZDX8G_ISA0_B4096_MAX_BG2 does not have alu-engine].[UNILOG][WARNING] xir::Op{name = Lstm_StandardLstmCell_layer_0_forward__it_cct_new, type = depthwise-fix} has been assigned to CPU: [DPUCZDX8G_ISA0_B4096_MAX_BG2 does not have alu-engine].[UNILOG][WARNING] xir::Op{name = Lstm_StandardLstmCell_layer_0_forward__h_next_new, type = depthwise-fix} has been assigned to CPU: [DPUCZDX8G_ISA0_B4096_MAX_BG2 does not have alu-engine].
[UNILOG][FATAL][XCOM_ITEM_UNDEFINED][The requested item is not finded or defined!]
*** Check failure stack trace: ***
Aborted`

I cannot figure out how to check further details about the error to figure out exactly what is going wrong.

Thank you

BraunPhilipp, 2021-08-20:
Are the layers supported by Xilinx as a default? If not I would work my way back from a reference LSTM example.

dongxuedx, 2021-08-23:
Looks like you're using the CNN compiler, if you want to compile  LSTM model, The RNN compiler is needed.

hanxue, 2022-03-17:
Hi @chtapodi ,

The RNN workflows are at https://github.com/Xilinx/Vitis-AI/tree/v2.0/demo/DPU-for-RNN . For RNN workflows, you will need to build the RNN docker https://github.com/Xilinx/Vitis-AI/tree/v2.0/demo/DPU-for-RNN

zhenzhen-AMD, 2022-07-22:
Hi @chtapodi ,
Has the problem been resolved? Do you have any other questions? If there are no more issues, we will close this issue. There is a problem later, please reopen the issue. Thank you.

dongxuedx, 2022-07-22:
> Resent-From: dzhy Xilinx ***@***.***>
>
> CAUTION: This message has originated from an External Source. Please use proper judgment and caution when opening attachments, clicking links, or responding to this email.

[AMD Official Use Only - General]

HI. You can close it. Thank you.

closed 2022-07-22 by qianglin-xlnx