Xilinx/Vitis-AI #509: Significance of NetName parameter and where to find netname for models in model zoo

daivikbhatia, 2021-08-12:
Hi,
I want to know what role netname plays in model performance and where can I find this parameter for models in model zoo?
Also What netname should I give If I plan to go for a custom model that's not present in Model Zoo.

Reason for these questions:
I have trained and deployed two models on my custom dataset, one is YoloV4 and other is Yolov4 tiny. The accuracy I got for Yolov4 was good but for Yolov4 tiny, it was messed up badly. 
As there is no netname for Yolov4 tiny, therefore I used the same netname as Yolov4. Can this be the reason for accuracy mess up? If yes, then what should I put in netname for yolov4 tiny. I'm using DNNDK's N2cube for some unavoidable reason for deployment and in that, Netname also goes for the Kernel name. 
Your answers will clear out a lot of confusions for me. 

Thankyou.

Mookel, 2021-08-17:
Hi, @daivikbhatia  what's the accuracy of Yolov4 tiny after quantization using DECENT?

If the quantized model itself has low accuracy, then the deployed accuracy will also be the same.

The netname has no effects on the performance, but must be consistent in the compilation srcipt and in the deploy c++ code, thus the runtime can load DPU kernels correctly. For net name usage purpose, you can refer to following UG for more details, thanks very much.

https://www.xilinx.com/support/documentation/sw_manuals/vitis_ai/1_2/ug1414-vitis-ai.pdf 

- page 77:
> --net_name Name of DPU kernel for network model after compiled by VAI_C.

- page 143:
> netName: The pointer to neural network name. Use the names produced by Deep Neural
Network Compiler (VAI_C) after the compilation of neural network. For each DL application,
perhaps there are many DPU Kernels existing in its hybrid CPU+DPU binary executable. For
each DPU Kernel, it has one unique name for differentiation purpose.

daivikbhatia, 2021-09-29:
Hi, thank you so much for your reply. It helped a lot. It turned out I was using wrong anchors for v4 tiny. I corrected that and got the expected accuracy.

closed 2021-09-29 by daivikbhatia