Xilinx/Vitis-AI #510: Docker fails to mount the Vitis-AI directory if it contains symlinks

nickfraser, 2021-08-12:
If the Vitis-AI directory contains symlinks, the following error occurs when running the `./docker_run.sh` script:

```
Do you agree to the terms and wish to proceed [y/n]? y
docker: Error response from daemon: error while creating mount source path '/home/nfraser/scratch/workspace/vitis_ai/Vitis-AI': mkdir /home/nfraser/scratch: permission denied.
```

hanxue, 2022-03-17:
Hi @nickfraser ,

Are you using the latest docker_run.sh from the `master` branch, or from the `v2.0` Vitis AI release? There are known issues with the older versions of docker_run.sh that has now been fixed: https://github.com/Xilinx/Vitis-AI/blob/v2.0/docker_run.sh 

You will also need to have write access to the directory (`$PWD`) where you are running docker_run.sh . From the error message, `/home/nfraser/scratch/workspace/vitis_ai/Vitis-AI` or `/home/nfraser/scratch` needs to have write access by the Linux user that is starting `docker_run.sh`

qianglin-xlnx, 2023-01-16:
Closing since no activity for more than half a year, please open a new issue if you still have any questions, thanks

closed 2023-01-16 by qianglin-xlnx