Xilinx/Vitis-AI #514: XVDPU build fails when BATCH_N=6

ksstms, 2021-08-13:
Hi! For some reason this piece of code:

https://github.com/Xilinx/Vitis-AI/blob/f61061eef7550d98bf02a171604c9a9f283a7c47/dsa/XVDPU-TRD/vitis_prj/scripts/makefile#L60-L64

produces these lines in `vitis_cfg.vh`:
```
-e `define wrp_UBANK_IMG_MRS       1
-e `define wrp_UBANK_WGT_MRS       1
-e `define wrp_ELEW_MULT_ENA       0
```

Of course the build exits with an error. Manually removing the `-e`  between `make files` and  `make all` solves the problem.

qianglin-xlnx, 2021-08-17:
Hi @ksstms 
Thank you for your feedback. We will fix it. Before we fix it, you can try the following :
 ```
ifeq  ($(BATCH_N), 6) 
 	printf "\`define wrp_UBANK_IMG_MRS       1\n" >> $(CFG_VH_PATH) 
 	printf "\`define wrp_UBANK_WGT_MRS       1\n" >> $(CFG_VH_PATH) 
 	printf "\`define wrp_ELEW_MULT_ENA       0\n" >> $(CFG_VH_PATH) 
 endif 
```
Or manually remove the `-e` just like you did.

ksstms, 2021-08-23:
Thanks. I have fixed it in the meantime and opened a pull request (#515), but forgot to mention it here.