Xilinx/Vitis-AI #517: Quantizer incorrectly determines shape for transposed convolution / deconvolution

BraunPhilipp, 2021-08-20:
The compilation error happens for the `center_frozen.pb` model. The model is derived from a Keras model `model.h5`. I have tested the model with the standard tensorflow interfaces without any errors. For testing purposes I have included `pb_test.py` and a test tensor in `000000_intermediate.npy`. According to PG367 by Xilinx deconvolutions are supported by the compiler. Quantization and compilation are both executed on the newest version of Vitis Ai (master branch). Commands used for quantization and compilation are included below. I also tested with a batch size of 4 due to a previous issue with the same outcome.

Note that I was able to extract the xir compiler frontend. The `xnnc-run` binary unfortunately seems to be closed source. The issue could be caused by the combination of concatenation and deconvolution.

@paolodalberto @woinck If you have any ideas what the issue you could be please let me know :)

**Quantization and compilation command**

```
vai_q_tensorflow quantize --input_frozen_graph=./output_new/center_frozen.pb --input_nodes=input_1 --input_shapes=1,504,504,64 --output_nodes=cnn/concatenate/concat --input_fn=calib_load.calib_input_intermediate --method=1 --gpu=0 --output_dir=./output_new_quantize

vai_c_tensorflow --frozen_pb="./output_new_quantize/quantize_eval_model.pb" --arch /opt/vitis_ai/compiler/arch/DPUCAHX8H/U280/arch.json --output_dir="output_new_quantize" --options="{'mode':'normal', 'input_shape': '1,504,504,64'}" --net_name "point_pillars" &> output_new_quantize/relu.log
```

**The debug folder contains the following files**

```
debug
├── 000000_intermediate.npy
├── center_frozen.pb
├── center.h5
├── center.pb
├── model.h5
├── pb_test.py
├── quantize_eval_model.pb
└── relu.log
```

**Compilation output**

```
[INFO] Namespace(batchsize=1, inputs_shape=['1,504,504,64'], layout='NHWC', model_files=['./output_new_quantize/quantize_eval_model.pb'], model_type='tensorflow', named_inputs_shape=None, out_filename='/tmp/point_pillars_org.xmodel', proto=None)
in_shapes: [[1, 504, 504, 64]]
[INFO] tensorflow model: /workspace/pointpillars/PointPillars/output_new_quantize/quantize_eval_model.pb

[INFO] parse raw model     :  0%|          | 0/61 [00:00<?, ?it/s]                           
[INFO] parse raw model     :100%|██████████| 61/61 [00:00<00:00, 29605.71it/s]               

[INFO] infer shape (NHWC)  :  0%|          | 0/64 [00:00<?, ?it/s]                           
[INFO] infer shape (NHWC)  : 27%|██▋       | 17/64 [00:00<00:00, 156.33it/s]                 
[INFO] infer shape (NHWC)  : 64%|██████▍   | 41/64 [00:00<00:00, 203.82it/s]                 
[INFO] infer shape (NHWC)  : 98%|█████████▊| 63/64 [00:00<00:00, 199.76it/s]                 
[INFO] infer shape (NHWC)  :100%|██████████| 64/64 [00:00<00:00, 199.00it/s]                 

[INFO] perform level-0 opt :  0%|          | 0/2 [00:00<?, ?it/s]                            
[INFO] perform level-0 opt :100%|██████████| 2/2 [00:00<00:00, 920.61it/s]                   

[INFO] perform level-1 opt :  0%|          | 0/6 [00:00<?, ?it/s]                            
[INFO] perform level-1 opt :100%|██████████| 6/6 [00:00<00:00, 963.40it/s]                   

[INFO] generate xmodel     :  0%|          | 0/61 [00:00<?, ?it/s]                           
[INFO] generate xmodel     :100%|██████████| 61/61 [00:00<00:00, 7913.78it/s]                
[INFO] dump xmodel ...
[INFO] dump xmodel: /tmp/point_pillars_org.xmodel
[UNILOG][INFO] Target architecture: DPUCAHX8H_ISA2
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCAHX8H_ISA2
[UNILOG][INFO] Graph name: quantize_eval_model, with op num: 135
[UNILOG][INFO] Begin to compile...
[0;31m[UNILOG][FATAL][XIR_SHAPE_UNMATCH][The shape is unmatching.] Using xir::Op{name = cnn/up3/conv2dt/conv2d_transpose, type = transposed-conv2d-fix} to replace xir::Op{name = tf_op_layer_Relu_3/Relu_3/aquant_cnn/concatenate/concat, type = fix2float}'s input xir::Op{name = tf_op_layer_Relu_3/Relu_3/aquant, type = float2fix} failed. Size is unmatching, the original input shape is {1, 252, 252, 128} but the new input shape is {1, 251, 251, 128}.
[m*** Check failure stack trace: ***
Aborted (core dumped)
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
```

BraunPhilipp, 2021-08-20:
**Attached debug folder**

[debug.zip](https://github.com/Xilinx/Vitis-AI/files/7021003/debug.zip)

paolodalberto, 2021-08-20:
I will take at the quantized model.  @BraunPhilipp 
just looking at the error log, I think I will be powerless  but I could provide a starting point

paolodalberto, 2021-08-20:
I cannot reproduce the same error ... but I Cannot compile it either.
something is off with one deconvolution

paolodalberto, 2021-08-21:
Currently I have found a synchronization bug for the first transpose. thank you.
The last transpose uses stride 4. This stride has not been tested and the code is incorrect.

paolodalberto, 2021-08-23:
@BraunPhilipp 
I will not be able to fix the unmatched dimensions  but I am asking who could.
The deconvolution you have stride=4, kernel=3, and padding 1. Currently I have two methodologies to handle this case and both fails for different reasons.  The optimized one (i.e., only convolutions) cannot generate the 9 1x1 kernels. The original one (i.e., upsample + convolution) fails because the upsample stride 4 (this is a limitation of the compiler). 

A practical solution ,if there is any, is is to have the 4 stride deconvolution be split into two 2-stride deconvolutions. Does it make sense to you ?

paolodalberto, 2021-08-23:
I think I can formulate a solution for the case stride > kernel ...
but this will not help you :(

paolodalberto, 2021-08-24:
> This is a known issue. For transposed conv, input_shape_w_h [63, 63], kernel = 3, stride = 4, pad_mode = "same", output_shape is 64 * 4, For DPU, output_shape = (63 - 1) * 4 + 1 + 3 - 1 - pad = 251 - pad, so the pad would be -1. pad = -1 is an undefined behavior now. Additionally, stride > kernel convolution is not supported by DPU now. Maybe we can define this computation pattern and allocate this operator to CPU.

paolodalberto, 2021-08-24:
I think the compiler will have means to deal with this case in 1.4.1
I will use this case to create tests set and verify the compiler solution: golden set ... 
My colleagues above suggested this is a know  problem for DPUs ...

BraunPhilipp, 2021-08-29:
I will try modify the deconvolution. Thank you for the support!

paolodalberto, 2021-08-30:
always

BraunPhilipp, 2021-09-05:
Split the deconvolution into two strided deconvolutions. Worked without issues :)

closed 2021-09-05 by BraunPhilipp

paolodalberto, 2021-09-29:
hurrah