Xilinx/Vitis-AI #518: vai_c_xir compile error with pytorch for resnet18

simplelins, 2021-08-23:
Hi, 
  I follow the steps of [pytorch](https://github.com/Xilinx/Vitis-AI/tools/Vitis-AI-Quantizer/vai_q_pytorch/README.md). everything is ok. 
  
[image]

  But when try to compile, some warning as '[UNILOG][WARNING] DPU prefers xir::Op{name = ResNet__ResNet_Sequential_layer4__BasicBlock_0__input_48, type = eltwise-fix}'s input batchto be 4, but it's 1 now. So it will be assigned to CPU'
  
[image]
I use the vitis-ai v1.4. the dpu subgraph number is 0, please help.
How to change the batch size  when compiling?

hoadv-qh, 2021-09-18:
Hi @simplelins,

I think you can refer to follow steps [here ](https://www.xilinx.com/support/documentation/sw_manuals/vitis_ai/1_3/ug1414-vitis-ai.pdf)(P.70).
Specially, pay attention to below.
<pre>
2. Check if there are modules to be called multiple times
</pre>

(above reference is for vitis-ai1.3, but you can find 1.4 version too)

qianglin-xlnx, 2022-08-24:
Hi @simplelins 
Closing since no activity for more than half year, please reopen if you still have question, thanks

closed 2022-08-24 by qianglin-xlnx