Xilinx/Vitis-AI #519: [vai 1.4 - oom error ]f2 fast fintune on custom resnet50

hill19850213, 2021-08-23:
Hi,

I have enabled the tf2 fast finetune on my custom resent50, here is my code.

`          model = vitis_quantize.VitisQuantizer(model).quantize_model(calib_dataset=quant_dataset,
                                                                  include_cle=True, 
                                                                  cle_steps=10,
                                                                  include_fast_ft=True,
                                                                  fast_ft_epochs=10)   `

I got an oom error as below. I also tried to adjust the fast_ft_epochs to other value but the oom error was still happened.
[image]

My VGA VRAM is 8 GB, but it seems it is not enough. Do you any recommend about oom?

xieyi4650, 2022-03-02:
try reduce `calib_batch_size` you can set it in vitis_quantize.VitisQuantizer(model).quantize_model

renjiangyong, 2022-09-06:
@hill19850213  Does OOM resolve? this issue is existed for a long time, I will close this issue, if you have any problem you can reopen it ,thank you .

closed 2022-09-06 by renjiangyong