Xilinx/Vitis-AI #520: [vai_q_tensorflow2.x] Supported layer raises an error upon quantization.

liunett, 2021-08-23:
Hi,

I am trying to implement a Multiply layer on a TensorFlow model but the following error raises when quantizing.
**Both Vitis-AI 1.4 documentation and my DPUCZDX8G docs, specify that tf.keras.layers.Multiply and Elementwise-Multiplication are supported.**

```---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-17-afd1aa630b22> in <module>
      1 from tensorflow_model_optimization.quantization.keras import vitis_quantize
      2 quantizer = vitis_quantize.VitisQuantizer(model)
----> 3 quantized_model = quantizer.quantize_model(calib_dataset=(x_data))

/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py in quantize_model(self, loss, metrics, calib_dataset, calib_batch_size, calib_steps, eval_dataset, verbose, configs, **kwargs)
    520       # Optimize model before quantization
    521       if not self._optimized_model:
--> 522         self.optimize_model()
    523 
    524       logger.debug('Quantize Pipeline Configurations:')

/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py in optimize_model(self, configs, **kwargs)
    411       self._quantize_strategy.get_optimize_pipeline().print_configs()
    412 
--> 413       self._create_optimized_model()
    414     return self._optimized_model
    415 

/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py in _create_optimized_model(self)
    210         candidate_layers=self._candidate_layers,
    211         layer_metadata=self._layer_metadata,
--> 212         quantize_strategy=self._quantize_strategy)
    213 
    214   def _create_analysed_model(self, dataset):

/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py in create_optimize_model(model, candidate_layers, layer_metadata, quantize_strategy)
    634   optimize_pipeline = quantize_strategy.get_optimize_pipeline()
    635   optimized_model, layer_metadata = optimize_pipeline.apply(
--> 636       model, candidate_layers, layer_metadata)
    637   return optimized_model, layer_metadata
    638 

/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/eight_bit/vitis_8bit_transforms_pipeline.py in apply(self, model, candidate_layers, layer_metadata)
     72 
     73     transformed_model, layer_metadata = _apply_availables(
---> 74         model, configs, available_transforms, candidate_layers, layer_metadata)
     75 
     76     # Train with bn is conflict with fold bn params

/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/eight_bit/vitis_8bit_transforms_pipeline.py in _apply_availables(model, configs, available_transforms, candidate_layers, layer_metadata)
     48   transformed_model, layer_metadata = model_transformer.ModelTransformer(
     49       model, transforms, candidate_layers,
---> 50       layer_metadata).recursive_transform()
     51   return transformed_model, layer_metadata
     52 

/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/graph_transformations/model_transformer.py in recursive_transform(self)
    631     return ModelTransformer(transformed_model, self.transforms,
    632                             self.candidate_layers,
--> 633                             self.layer_metadata).transform()

/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/graph_transformations/model_transformer.py in transform(self)
    573           # For Transforms, that may inadvertently do this we can end up in
    574           # an infinite loop. Move on if no meaningful change has been made.
--> 575           if match_layer_node == replacement_layer_node:
    576             continue
    577 

/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/graph_transformations/transforms.py in __eq__(self, other)
    170       return False
    171 
--> 172     if self.layer != other.layer \
    173         or not self._eq(self.weights, other.weights) \
    174         or self.metadata != other.metadata:

ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()
```

I have tried several models even the simplest one fails.

```
import numpy as np
import tensorflow as tf
from tensorflow import keras
from tensorflow.keras import layers

x_data = np.empty((2,2), dtype=np.float32)
mult = np.empty((2,2), dtype=np.float32)

inputs = layers.Input(shape=(2,2))
outputs = layers.Multiply()([inputs, mult])

model = tf.keras.Model(inputs=inputs, outputs=outputs)

from tensorflow_model_optimization.quantization.keras import vitis_quantize
quantizer = vitis_quantize.VitisQuantizer(model)
quantized_model = quantizer.quantize_model(calib_dataset=(x_data))
```

xieyi4650, 2022-03-09:
Hi @liunett 
I have not reproduce the error using the code you provide, 
But  I got a error below, have you encounter this error?

`Traceback (most recent call last):
  File "test.py", line 22, in <module>
    quantized_model = quantizer.quantize_model(calib_dataset=(x_data))
  File "/proj/rdi/staff/xieyi/anaconda3/envs/tf2.6/lib/python3.6/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 567, in quantize_model
    
  File "/proj/rdi/staff/xieyi/anaconda3/envs/tf2.6/lib/python3.6/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 455, in optimize_model
    return self._optimized_model
  File "/proj/rdi/staff/xieyi/anaconda3/envs/tf2.6/lib/python3.6/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 229, in _create_optimized_model
    
  File "/proj/rdi/staff/xieyi/anaconda3/envs/tf2.6/lib/python3.6/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 701, in create_optimize_model
    model, candidate_layers, layer_metadata)
  File "/proj/rdi/staff/xieyi/anaconda3/envs/tf2.6/lib/python3.6/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/eight_bit/vitis_8bit_transforms_pipeline.py", line 74, in apply
    model, configs, available_transforms, candidate_layers, layer_metadata)
  File "/proj/rdi/staff/xieyi/anaconda3/envs/tf2.6/lib/python3.6/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/eight_bit/vitis_8bit_transforms_pipeline.py", line 50, in _apply_availables
    layer_metadata).recursive_transform()
  File "/proj/rdi/staff/xieyi/anaconda3/envs/tf2.6/lib/python3.6/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/graph_transformations/model_transformer.py", line 742, in recursive_transform
    self.model, clone_function=_make_transform_fn())
  File "/proj/rdi/staff/xieyi/anaconda3/envs/tf2.6/lib/python3.6/site-packages/keras/models.py", line 452, in clone_model
    model, input_tensors=input_tensors, layer_fn=clone_function)
  File "/proj/rdi/staff/xieyi/anaconda3/envs/tf2.6/lib/python3.6/site-packages/keras/models.py", line 196, in _clone_functional_model
    created_layers=created_layers))
  File "/proj/rdi/staff/xieyi/anaconda3/envs/tf2.6/lib/python3.6/site-packages/keras/engine/functional.py", line 1283, in reconstruct_from_config
    process_node(layer, node_data)
  File "/proj/rdi/staff/xieyi/anaconda3/envs/tf2.6/lib/python3.6/site-packages/keras/engine/functional.py", line 1231, in process_node
    output_tensors = layer(input_tensors, **kwargs)
  File "/proj/rdi/staff/xieyi/anaconda3/envs/tf2.6/lib/python3.6/site-packages/keras/engine/base_layer.py", line 978, in __call__
    input_list)
  File "/proj/rdi/staff/xieyi/anaconda3/envs/tf2.6/lib/python3.6/site-packages/keras/engine/base_layer.py", line 1116, in _functional_construction_call
    inputs, input_masks, args, kwargs)
  File "/proj/rdi/staff/xieyi/anaconda3/envs/tf2.6/lib/python3.6/site-packages/keras/engine/base_layer.py", line 848, in _keras_tensor_symbolic_call
    return self._infer_output_signature(inputs, args, kwargs, input_masks)
  File "/proj/rdi/staff/xieyi/anaconda3/envs/tf2.6/lib/python3.6/site-packages/keras/engine/base_layer.py", line 889, in _infer_output_signature
    outputs = call_fn(inputs, *args, **kwargs)
  File "/proj/rdi/staff/xieyi/anaconda3/envs/tf2.6/lib/python3.6/site-packages/keras/layers/merge.py", line 121, in call
    input_ndims = list(map(backend.ndim, inputs))
  File "/proj/rdi/staff/xieyi/anaconda3/envs/tf2.6/lib/python3.6/site-packages/keras/backend.py", line 1449, in ndim
    return x.shape.rank
AttributeError: 'list' object has no attribute 'shape'
`

zhenzhen-AMD, 2023-07-07:
Hi @rubenlesmes ,

The usage of tf.keras.layers.Multiply in the example you provided is incorrect. 
The inputs to the Multiply layer must be tensors. 
However, in your provided example, the data type of 'inputs' is a **tensor**, while 'mult' is a **NumPy array**. 
Therefore, an error will occur during the quantization stage.

Here is a successful example of quantization.

`
import numpy as np
import tensorflow as tf

x_train = np.random.random((100, 224, 224, 3))
y_train = np.random.random((100, 224, 224, 3))

input1 = tf.keras.layers.Input(shape=(224, 224, 3))
input2 = tf.keras.layers.Input(shape=(224, 224, 3))
output = tf.keras.layers.Multiply()([input1, input2])
model = tf.keras.models.Model(inputs=[input1, input2], outputs=output)

from tensorflow_model_optimization.quantization.keras import vitis_quantize
quantizer = vitis_quantize.VitisQuantizer(model)
quantized_model = quantizer.quantize_model(calib_dataset=(x_train, y_train))
`

I hope this helps! Let me know if you have any further questions.

zhenzhen-AMD, 2023-07-19:
Closing this for now. Please re-open if there are any other concerns. Thanks.

closed 2023-07-19 by renjiangyong