Xilinx/Vitis-AI #522: Calling vart.Runner.create_runner(...) multiple times leads to wrong input shape

BraunPhilipp, 2021-08-29:
I am currently facing a really curious issue. I have a model which is used to derive multiple subgraphs. Calling the DPU runner on these subgraphs leads to an incorrect first dimension (batch size on the second and following subgraphs). Exactly one dimension is added to the tensor. The first subgraph has the correct input size.

`dpu_runner = vart.Runner.create_runner(subgraph, "run")
input_tensors = dpu.get_input_tensors()
`

**So instead of (4, 504, 504, 64) I get a batch size of (5, 504, 504, 64).**

I am running on a recent version of Vitis AI. I hope someone can look into this.

BraunPhilipp, 2021-08-29:
Sorry for bothering again. I keep running into bugs in Vitis AI. @paolodalberto 

If you have any suggestions please let me know. First subgraph always has the correct shape. Ordering has no impact. Resetting the FPGA also has no impact. First dimension is incorrect starting at the second DPU runner.

qianglin-xlnx, 2021-08-30:
@BraunPhilipp 
Which platform are you running the model?

BraunPhilipp, 2021-08-30:
I am running on a recent U280 with the DPUCAHX8H. I am not facing any issues during quantization and compilation.

paolodalberto, 2021-08-30:
@BraunPhilipp this is strange.
The compiler does not and cannot change the batch size (or tensor shape) strictly forbidden.
It sounds more like a partitioning fluke  or a shape computation when partitioning. 

If you can provide a quantize model I can always try to compile and, with some help, see if we ca reproduce it.

BraunPhilipp, 2021-09-05:
In the following folder are the protobufs for the models I am quantizing and the compiler output. 

[center_frozen.zip](https://github.com/Xilinx/Vitis-AI/files/7112277/center_frozen.zip)

@paolodalberto My current assumption is that the VART API is incorrectly determining the input shape.

[https://www.xilinx.com/html_docs/xilinx2019_2/vitis_doc/tpa1605601589077.html](https://www.xilinx.com/html_docs/xilinx2019_2/vitis_doc/tpa1605601589077.html
)

gaoyue0712Xlinx, 2021-09-10:
@BraunPhilipp  try to use this function to create runners  std::unique_ptr<RunnerExt> create_runner(const xir::Subgraph* subgraph, xir::Attrs* attrs); 
please use same attrs if you want to run with same batch  when running on DPUCAHX8H. Here is an example https://github.com/Xilinx/Vitis-AI/blob/master/demo/VART/resnet50_ext/resnet50.cpp#L126-128

BraunPhilipp, 2021-09-16:
The issue was likely caused by the DPU configuration. DPUCAHX8H integrates a 4ENGINE and 5ENGINE design.

closed 2021-09-16 by BraunPhilipp