Xilinx/Vitis-AI #523: DPU-TRD Kernel image boot failure (Vitis flow)

likecaulfield, 2021-08-30:
Dear community,

Hello I'm currently working on building a kernel image following DPU-TRD Vitis flow: https://github.com/Xilinx/Vitis-AI/blob/1.3.1/dsa/DPU-TRD/prj/Vitis/README.md

 

I modified some parameters for zcu106 board as it has less BRAMs and I had to build the required files such as .xsa, .xpfm as there is no pre-built ones for zcu106 board. (But It doesn't seem this is the reason)

 

As you can see in the log, everything seems fine until building the kernel image but when I flash the image (sd_card.img) using balenaEtcher and tried to boot with gtkterm, nothing shows up.

[image]

I've attached the full log. It is a bit more confusing because there is no any error message and it is still not booting.

 

It would be really appreciated if anyone can give me some advice on this error.

 

Thank you.

 

Best regards,

 

Joon

joon@joon-X299X-DESIGNARE-10G:~/dpu_prj/ttrl3/pfm3/DPU-TRD/prj/Vitis$ export EDGE_COMMON_SW=/home/joon/dpu_prj/ttrl3/pfm3/xilinx-zynqmp-common-v2020.2
joon@joon-X299X-DESIGNARE-10G:~/dpu_prj/ttrl3/pfm3/DPU-TRD/prj/Vitis$ export SDX_PLATFORM=/home/joon/dpu_prj/ttrl3/pfm3/wksp3/zcu106_B512_3/export/zcu106_B512_3/zcu106_B512_3.xpfm
joon@joon-X299X-DESIGNARE-10G:~/dpu_prj/ttrl3/pfm3/DPU-TRD/prj/Vitis$ make KERNEL=DPU_SM DEVICE=zcu106
/home/joon/Xilinx/Vivado/2020.2/bin/vivado -mode batch -source /home/joon/dpu_prj/ttrl3/pfm3/DPU-TRD/prj/Vitis/scripts/gen_dpu_xo.tcl -notrace -tclargs binary_container_1/dpu.xo DPUCZDX8G hw zcu106

****** Vivado v2020.2 (64-bit)
**** SW Build 3064766 on Wed Nov 18 09:12:47 MST 2020
**** IP Build 3064653 on Wed Nov 18 14:17:31 MST 2020
** Copyright 1986-2020 Xilinx, Inc. All Rights Reserved.

source /home/joon/dpu_prj/ttrl3/pfm3/DPU-TRD/prj/Vitis/scripts/gen_dpu_xo.tcl -notrace
binary_container_1/dpu.xo
INFO: [IP_Flow 19-5654] Module 'DPUCZDX8G' uses SystemVerilog sources with a Verilog top file. These SystemVerilog files will not be analysed by the packager.
INFO: [IP_Flow 19-1842] HDL Parser: Found include file "src/arch_def.vh" from the top-level HDL file.
INFO: [IP_Flow 19-1842] HDL Parser: Found include file "/home/joon/dpu_prj/ttrl3/pfm3/DPU-TRD/prj/Vitis/dpu_conf.vh" from the top-level HDL file.
INFO: [IP_Flow 19-1841] HDL Parser: Add include file "/home/joon/dpu_prj/ttrl3/pfm3/DPU-TRD/prj/Vitis/dpu_conf.vh" to file group xilinx_anylanguagesynthesis.
INFO: [IP_Flow 19-1841] HDL Parser: Add include file "/home/joon/dpu_prj/ttrl3/pfm3/DPU-TRD/prj/Vitis/dpu_conf.vh" to file group xilinx_anylanguagebehavioralsimulation.
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/home/joon/Xilinx/Vivado/2020.2/data/ip'.
INFO: [IP_Flow 19-5107] Inferred bus interface 'aclk' of definition 'xilinx.com:signal:clock:1.0' (from X_INTERFACE_INFO parameter from HDL file).
INFO: [IP_Flow 19-5107] Inferred bus interface 'aclk' of definition 'xilinx.com:signal:clock:1.0' (from 'X_INTERFACE_INFO' attribute).
INFO: [IP_Flow 19-5107] Inferred bus interface 'ap_clk_2' of definition 'xilinx.com:signal:clock:1.0' (from X_INTERFACE_INFO parameter from HDL file).
INFO: [IP_Flow 19-5107] Inferred bus interface 'ap_clk_2' of definition 'xilinx.com:signal:clock:1.0' (from 'X_INTERFACE_INFO' attribute).
INFO: [IP_Flow 19-5107] Inferred bus interface 'ap_rst_n_2' of definition 'xilinx.com:signal:reset:1.0' (from X_INTERFACE_INFO parameter from HDL file).
INFO: [IP_Flow 19-5107] Inferred bus interface 'ap_rst_n_2' of definition 'xilinx.com:signal:reset:1.0' (from 'X_INTERFACE_INFO' attribute).
INFO: [IP_Flow 19-5107] Inferred bus interface 'aresetn' of definition 'xilinx.com:signal:reset:1.0' (from X_INTERFACE_INFO parameter from HDL file).
INFO: [IP_Flow 19-5107] Inferred bus interface 'aresetn' of definition 'xilinx.com:signal:reset:1.0' (from 'X_INTERFACE_INFO' attribute).
INFO: [IP_Flow 19-5107] Inferred bus interface 'M_AXI_GP0' of definition 'xilinx.com:interface:aximm:1.0' (from Xilinx Repository).
INFO: [IP_Flow 19-5107] Inferred bus interface 'M_AXI_HP0' of definition 'xilinx.com:interface:aximm:1.0' (from Xilinx Repository).
INFO: [IP_Flow 19-5107] Inferred bus interface 'M_AXI_HP2' of definition 'xilinx.com:interface:aximm:1.0' (from Xilinx Repository).
INFO: [IP_Flow 19-5107] Inferred bus interface 'S_AXI_CONTROL' of definition 'xilinx.com:interface:aximm:1.0' (from Xilinx Repository).
INFO: [IP_Flow 19-5107] Inferred bus interface 'interrupt' of definition 'xilinx.com:signal:interrupt:1.0' (from Xilinx Repository).
INFO: [IP_Flow 19-4728] Bus Interface 'interrupt': Added interface parameter 'SENSITIVITY' with value 'LEVEL_HIGH'.
INFO: [IP_Flow 19-4728] Bus Interface 'aclk': Added interface parameter 'ASSOCIATED_BUSIF' with value 'M_AXI_GP0'.
INFO: [IP_Flow 19-4728] Bus Interface 'aclk': Added interface parameter 'ASSOCIATED_RESET' with value 'aresetn'.
INFO: [IP_Flow 19-4728] Bus Interface 'ap_clk_2': Added interface parameter 'ASSOCIATED_RESET' with value 'ap_rst_n_2'.
INFO: [IP_Flow 19-4728] Bus Interface 'ap_rst_n_2': Added interface parameter 'POLARITY' with value 'ACTIVE_LOW'.
[... 420 lines cut ...]
sd_card_ext4_start:2000896
dummy_ext4_start:6834000
2096577+0 records in
2096577+0 records out
1073447424 bytes (1.1 GB, 1.0 GiB) copied, 2.28127 s, 471 MB/s
4833104+0 records in
4833104+0 records out
2474549248 bytes (2.5 GB, 2.3 GiB) copied, 5.22234 s, 474 MB/s
409776+0 records in
409776+0 records out
209805312 bytes (210 MB, 200 MiB) copied, 0.404818 s, 518 MB/s

Section: 'SYSTEM_METADATA'(22) was successfully added.
Size : 20071 bytes
Format : RAW
File : '/home/joon/dpu_prj/ttrl3/pfm3/DPU-TRD/prj/Vitis/_x/package/packagedSystemDiagram.json'
Successfully wrote (13691388 bytes) to the output file: /home/joon/dpu_prj/ttrl3/pfm3/DPU-TRD/prj/Vitis/_x/package/upsert.xclbin
INFO: [v++ 60-2460] Successfully copied a temporary xclbin to the output xclbin: /home/joon/dpu_prj/ttrl3/pfm3/DPU-TRD/prj/Vitis/dpu.xclbin
INFO: [v++ 60-2343] Use the vitis_analyzer tool to visualize and navigate the relevant reports. Run the following command.
vitis_analyzer /home/joon/dpu_prj/ttrl3/pfm3/DPU-TRD/prj/Vitis/dpu.xclbin.package_summary
INFO: [v++ 60-791] Total elapsed time: 0h 14m 28s
INFO: [v++ 60-1653] Closing dispatch client.
cp ./binary_*/link/vivado/vpl/prj/prj*/sources_1/bd/*/hw_handoff/*.hwh ./binary_*/sd_card
cp ./binary_*/link/vivado/vpl/prj/prj.gen/sources_1/bd/*/ip/*_DPUCZDX8G_1_0/arch.json ./binary_*/sd_card
joon@joon-X299X-DESIGNARE-10G:~/dpu_prj/ttrl3/pfm3/DPU-TRD/prj/Vitis$

likecaulfield, 2021-08-30:
Please let me know if any log file required

qianglin-xlnx, 2021-08-30:
Hi @likecaulfield 
Could you boot up successfully with the SD card image without integrating the DPU?

likecaulfield, 2021-08-31:
Hi @qianglin-xlnx

I tried to boot with these three files but I failed.

BOOT.BIN, boot.scr, image.ub  (from ~images/linux)

I copied it into sd card and tried to boot with gtkterm but still nothing shows up.

Is it to do with software image build then?

Thank you.

likecaulfield, 2021-09-02:
*update2

Hi @qianglin-xlnx 

I've found out that I might have made a mistake.

I built xsa file with DPU on Vivado and did integrate dpu on Vitis again.

Could it be the reason? I will try with no dpu integration on vivado.

Thank you.

closed 2021-09-09 by likecaulfield